jkjung-avt/tensorrt_demos #3: failed to open camera....

i-chaochen, 2019-07-19:
is any fig setting wrong?
```
$ python3 camera_trt_googlenet.py --usb --vid 0 --width 1280 --height 720
Called with args:
Namespace(crop_center=False, image_height=720, image_width=1280, rtsp_latency=200, rtsp_uri=None, use_rtsp=False, use_usb=True, video_dev=0)
Failed to open camera!
```

I changed the following one is working fine thought.

```
def open_cam_usb(dev, width, height):
    gst_str = ('v4l2src device=/dev/video{} ! '
               'video/x-raw, width=(int){}, height=(int){} ! '
               'videoconvert ! appsink').format(dev, width, height)
    # return cv2.VideoCapture(gst_str, cv2.CAP_GSTREAMER)
    return cv2.VideoCapture(0)
```

jkjung-avt, 2019-07-20:
This depends on whether gstreamer's "v4l2src" is able to handle the camera device you are using.  In other words, it's a compatibility issue between the camera and the gstreamer element.  When you do "cv2.VideoCapture(0)", it's leveraging opencv's internal mechanism (probably "libv4l") to handle the camera instead.  That's OK too.

I will keep my implementations since I'd like to be able to set different image resolutions, e.g. 1920x1080 or 1280x720, using the gstreamer pipeline.

gzz1529657064, 2019-10-14:
@jkjung-avt Is there a way to solve this compatibility problem? I want to use gstream.

jkjung-avt, 2019-10-14:
Check out 'v4l2src' documentation, and try to find a gstreamer pipeline which works for your webcam.  If you manage to do that, then you could modify the 'open_cam_usb()' function accordingly.

https://gstreamer.freedesktop.org/documentation/video4linux2/v4l2src.html?gi-language=c

Using 'v4l2-ctl' to check all color/resolution formats your webcam supports would also help.

https://superuser.com/questions/639738/how-can-i-list-the-available-video-modes-for-a-usb-webcam-in-linux

zychen2016, 2019-11-04:
could use MIPI CSI camera？Like Raspberry Pi Camera v2 in Jetson nano

jkjung-avt, 2019-12-15:
Yes, please refer to this post on my blog.

http://disq.us/p/22rtcid

closed 2019-12-15 by jkjung-avt

Tetsujinfr, 2020-09-01:
Great repo, thanks!
for those who want to use with a ZED camera (like me):
1) install the [gstreamer plugin for the ZED](https://github.com/stereolabs/zed-gstreamer)
2) in the file /tensorrt_demos/utils/camera.py replace:

> gst_str = ('v4l2src device=/dev/video{} ! video/x-raw, width=(int){}, height=(int){} ! videoconvert ! appsink').format(dev, width, height) 
with:
> gst_str = ('zedsrc stream-type=1 camera-resolution=3 camera-fps=100 ! autovideoconvert ! appsink').format(dev, width, height)

you can tweak the zedsrc source with the parameters mentioned on the zed gstreamer plugin repo readme (e.g. camera-resolution = 0, 1, 2, 3 for resp. 2K, HD, 720p, WGA)

Then execute the demo like with any USB cam, e.g. under /tensorrt_demos/:
`python3 trt_yolo.py --usb 0 -m yolov4-416`

Hope this will help someone. ciao

jkjung-avt, 2020-09-02:
@Tetsujinfr  Thanks for sharing this (use of ZED camera).