jkjung-avt/tensorrt_demos #4: Possible to use other caffeemodel ?

Kmarconi, 2019-08-01:
Hi, first, thank you for your amazing work, got 15 ms of inference time on my nano, really suprised about it. Was wondering if I could use any other caffeemodel than the one you tell to use ? I would like to recognize  stuff like person, cycle or suitcase for exemple.

Best regards !

jkjung-avt, 2019-08-02:
Sure, it should be easy to modify my code example to optimize another caffemodel, as long as the model takes image inputs and use NN layers natively supported by TensorRT (don't require plugins).

Kmarconi, 2019-08-02:
Can you guide me a little more in this ? Your ".engine" is little bit secret for me, and I would like to use something like ssd mobilenet coco caffeemodel in your program ? How would you do this ?

jkjung-avt, 2019-08-02:
The ".engine" file stores the NN model which is optimized by TensorRT.  In addition, a big portion of the "create_engine.cpp" code just follows TensorRT's [sampleGoogleNet](https://docs.nvidia.com/deeplearning/sdk/tensorrt-sample-support-guide/index.html#googlenet_sample) example.  If you'd like to know more about it, please read NVIDIA's documentation.

As to your question about optimizing a ssd_mobilenet_coco caffemodel, unfortunately it won't work right out of the bat.  There are NN layers in ssd models which are not directly supported by TensorRT.  You'll have to implement it with "plugins".

closed 2019-12-15 by jkjung-avt