jkjung-avt/tensorrt_demos #7: Running MTCNN with tensorrt

tranmanhdat, 2019-10-01:
I have some confusion here.
I run MTCNN without tensorrt by demo in here : 
[](https://github.com/AaronFan1992/jetson_nano_demo)
and compare with your code for MTCNN with tensorrt
I realize when no tensorrt FPS is higher,I dont understand why.

jkjung-avt, 2019-10-02:
Thanks for letting me know.  Could you share more information about your test case?  I will look into it.

* Are you testing this on a Jetson Nano or TX2 or else? Which TensorRT version is used?
* How large (what resolution: 1280x720 or else) is the input image or video?
* Roughly how many (ground-truth) faces are there in the test image?
* How much slower is my code comparing to[ jetson_nano_demo](https://github.com/AaronFan1992/jetson_nano_demo)?

tranmanhdat, 2019-10-02:
Im testing on Jetson Nano
I find out i did not change the min face size LOL,when i change it FPS is nearly same in mtcnn with tensorrt or not

jkjung-avt, 2019-10-02:
I do have a few tricks which could speed up MTCNN.  I haven't found time to implement them into the code.

I'd appreciate you sharing with me about your test case: image resolution and how many faces, etc.  It could help me to make better design decisions later on.

Thanks.

tranmanhdat, 2019-10-02:
I tested with resolution 1280x720 and 0,1,2 faces cases,i will test more and share result
I'm waiting for your implement.

closed 2019-10-02 by tranmanhdat

jkjung-avt, 2019-10-03:
Thanks for sharing the information.  I'll try to implement some improvements soon.

jkjung-avt, 2019-10-05:
@tranmanhdat  I have optimized the TrtPNet code.  Comparing to my Sep. 30 code, the current version runs over 30% faster overall.  For example, FPS number improved from 5.15 to 6.94 when I test with the same Avengers picture on Jetson Nano.

Please check out my latest code, recompile 'create_engines.cpp' and rebuild the TensorRT engine files.  Then run your test again.

I plan to write a blog post about how I implement the optimization soon.

tranmanhdat, 2019-10-05:
thank for sharing,i wil test on my case soon.
could you explain why the current version run 30% over faster,do you change some arguments?

jkjung-avt, 2019-10-05:
Please read my new blog post, [Optimizing TensorRT MTCNN](https://jkjung-avt.github.io/optimize-mtcnn/), for details.

tranmanhdat, 2019-10-06:
thank for your optimization!
I tested on my computer for your last update and i realized that without tensorRT it's run faster,the result FPS you can see in 2 files i attach.
i config with the same arguments,i used a large video but i was select a ROI ( 750x1000) for testing.
i used scale_factor=709,min_face_size=90 and set the max_batch=3.
[FPS_withoutTensorRT.txt](https://github.com/jkjung-avt/tensorrt_demos/files/3694613/FPS_withoutTensorRT.txt)
[FPS_withTensorRT.txt](https://github.com/jkjung-avt/tensorrt_demos/files/3694615/FPS_withTensorRT.txt)

jkjung-avt, 2019-10-07:
Thanks for sharing the test results with me.  I don't plan to do further optimization on my code for now.  But allow me to make a couple of comments:

1. My code was optimized for 1280x720 image inputs.  For 750x1000 images, my code would need to do rescaling (additional computation) on every image.  In addition, there would be padded zero pixels on the rght-hand side (for PNet) which are wasted.  To get the best performance with my implementation, you could modify input blob dimension in [det1_relu.prototxt](https://github.com/jkjung-avt/tensorrt_demos/blob/master/mtcnn/det1_relu.prototxt#L26), as well as the corresponding code in `utils/mtcnn.py`.  (Refer to my "Optimizing TensorRT MTCNN" blog post for details.)

2. In the TensorFlow (non-TensorRT) case, it seems that FPS number fluctuated (34~69) significantly during your test.  Do you know why?

tranmanhdat, 2019-10-08:
I changed image to 1280x720 but with tensorRT still lower without tensorRT,
i can't define why FPS increase/decrease so fastly and unstable,i realize that when image have faces FPS will decrease ( base on number of faces in image) and it will increase fast when image dont have any face

jkjung-avt, 2019-10-08:
Thanks again for sharing the result.

tranmanhdat, 2019-10-08:
your implemence show a better performance when have many faces in iamge ( about 3,4 and above) ! it's greate!