jkjung-avt/tensorrt_demos #8: GStreamer: unable to start pipeline!

gzz1529657064, 2019-10-04:
I read your blog"How to Capture and Display Camera Video with Python on Jetson TX2"
run command error!
"python3 tegra-cam.py --usb --vid 1 --width 1280 --height 720"

Called with args:
Namespace(image_height=720, image_width=1280, rtsp_latency=200, rtsp_uri=None, use_rtsp=False, use_usb=True, video_dev=1)
OpenCV version: 3.4.0
OpenCV Error: Unspecified error (GStreamer: unable to start pipeline
) in cvCaptureFromCAM_GStreamer, file /home/nvidia/src/opencv-3.4.0/modules/videoio/src/cap_gstreamer.cpp, line 890
VIDEOIO(cvCreateCapture_GStreamer (CV_CAP_GSTREAMER_FILE, filename)): raised OpenCV exception:

/home/nvidia/src/opencv-3.4.0/modules/videoio/src/cap_gstreamer.cpp:890: error: (-2) GStreamer: unable to start pipeline
 in function cvCaptureFromCAM_GStreamer

Failed to open camera!

jkjung-avt, 2019-10-04:
Maybe GStreamer's v4l2src does not work for the USB webcam you're using.  Please try if replace open_cam_usb() function with the following works.

```
def open_cam_usb(dev, width, height):
    return cv2.VideoCapture(dev)
```

gzz1529657064, 2019-10-05:
> Maybe GStreamer's v4l2src does not work for the USB webcam you're using. Please try if replace open_cam_usb() function with the following works.
> 
> ```
> def open_cam_usb(dev, width, height):
>     return cv2.VideoCapture(dev)
> ```

thank you replay, you suggestion works!

gzz1529657064, 2019-10-05:
I read your github repository 'tf_trt_models'.  I want to transform the mobilenet_v1_yolo_v3 tensorflow model, but it need a mobilenet_v1_yolo_v3_egohands.config. I don't know how to write this file. 
What should I do?

I feel that it is not easy to accelerate the YOLOv3  tensorflow model with tensorRT. I'm going to try caffe on Jetson TX2。Your blog has helped me a lot。Thank you very much!

jkjung-avt, 2019-10-05:
I think you need to train your own 'mobilenet_v1_yolo_v3_egohands.config' model.  You could refer to my [Training a Hand Detector with TensorFlow Object Detection API](https://jkjung-avt.github.io/hand-detection-tutorial/) blog post.  I demonstrated how to train SSD and Faster RCNN models there.

Since this question is not directly related to 'tensorrt_demos', I'd appreciate it if you post further comments/questions in the more proper places, e.g. Disqus forum on my blog post or my GitHub [hand-detection-tutorial](https://github.com/jkjung-avt/hand-detection-tutorial) repository.

closed 2019-10-05 by jkjung-avt