jkjung-avt/tensorrt_demos #18: FasterRCNN possible? [Potential Future Dev]

mbufi, 2019-11-05:
Hello Jk!

Like usual, awesome work on this repo. I am always learning tons from the work you create.

I have tried to replicate what you did with SSD obj det in this repo but for FasterRCNN... but am not having much luck.. Have you tried to use any Tensorflow object detection API Faster RCNN models with the new Python TensorRT Api? 

Please let me know if you were successful or if there is a plan to include this in the next addition to the repo. 

Thank you for all your hard work :smile:

jkjung-avt, 2019-11-05:
I might find time to work on YOLOv3 and Faster RCNN later.  But there's no firm timeline.

mbufi, 2019-11-06:
That would be amazing. Thanks for the update.

closed 2019-11-06 by mbufi