jkjung-avt/tensorrt_demos #21: build ssd engine failed

YoungSharp, 2019-11-20:
[TensorRT] ERROR: Could not register plugin creator:  FlattenConcat_TRT in namespace: 
WARNING: To create TensorRT plugin nodes, please use the `create_plugin_node` function instead.
NOTE: UFF has been tested with TensorFlow 1.14.0.
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
UFF Version 0.6.5
=== Automatically deduced input nodes ===
[name: "Input"
op: "Placeholder"
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: 1
      }
      dim {
        size: 3
      }
      dim {
        size: 300
      }
      dim {
        size: 300
      }
    }
  }
}
]
=========================================

Using output node NMS
Converting to UFF graph
Warning: No conversion function registered for layer: NMS_TRT yet.
Converting NMS as custom op: NMS_TRT
Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_conf as custom op: FlattenConcat_TRT
Traceback (most recent call last):
  File "build_engine.py", line 204, in <module>
    main()
  File "build_engine.py", line 187, in main
    debug_mode=DEBUG_UFF)
  File "/usr/local/lib/python3.6/site-packages/uff/converters/tensorflow/conversion_helpers.py", line 178, in from_tensorflow
    debug_mode=debug_mode)
  File "/usr/local/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py", line 94, in convert_tf2uff_graph
    uff_graph, input_replacements, debug_mode=debug_mode)
  File "/usr/local/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py", line 79, in convert_tf2uff_node
    op, name, tf_node, inputs, uff_graph, tf_nodes=tf_nodes, debug_mode=debug_mode)
  File "/usr/local/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py", line 47, in convert_layer
    return cls.registry_[op](name, tf_node, inputs, uff_graph, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/uff/converters/tensorflow/converter_functions.py", line 21, in convert_placeholder
    dtype = tf2uff.convert_tf2numpy_dtype(tf_node.attr['dtype'].type)
  File "/usr/local/lib/python3.6/site-packages/uff/converters/tensorflow/converter.py", line 103, in convert_tf2numpy_dtype
    return tf.as_dtype(dtype).as_numpy_dtype
  File "/usr/local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py", line 712, in as_dtype
    raise TypeError("Cannot convert value %r to a TensorFlow DType." % type_value)
TypeError: Cannot convert value 0 to a TensorFlow DType.

Tensorrt 6.0.1
cuda10.0
linux 16.04
command line: python3 build_engine.py  ssd_mobilenet_v2_egohands

YoungSharp, 2019-11-20:
plus --- system environment:
linux 16.04
cuda10.0
cudnn7.6.5
Tensorrt 6.0.1.5
tensorflow 1.12
python 3.6

the error is appeared by using ssd sample.

jkjung-avt, 2019-11-20:
I haven't tested the code with TensorRT 6.x before...

Are you running the code on a Jetson platform? Or a x86 PC?

YoungSharp, 2019-11-20:
on x86 PC

YoungSharp, 2019-11-20:
error happend when converting Input node.

jkjung-avt, 2019-11-20:
I don't have an answer right now.  My code was originally developed and tested on Jetson (arm64) platforms.

I might find time to test that later.

YoungSharp, 2019-11-20:
i will keep trying, thanks.

YoungSharp, 2019-11-22:
@jkjung-avt  
Do you mind to share your training code with?  
The code to train ssd_mobilenet_v2_egohands.pb
my code is latest "https://github.com/tensorflow/models/blob/master/research", and my model is at  "https://drive.google.com/file/d/1e-9cb_sLmuevA4h1IPP029vUH3IxlIx1/view?usp=sharing"
my email address is 331870428@qq.com
Appreciated.

jkjung-avt, 2019-11-22:
I already shared all my training code on GitHub.  I was using a slightly older snapshot of TensorFlow Object Detection API.  Please refer to README.md in the repository and the associated blog posts for details.

https://github.com/jkjung-avt/hand-detection-tutorial

YoungSharp, 2019-11-22:
Thanks really

YoungSharp, 2019-11-22:
Thanks really

jkjung-avt, 2019-11-22:
By the way, the problem you're facing might be due to some changes in TensorRT 6.0.  I think the problem would likely go away if use TensorRT 5.x instead.

YoungSharp, 2019-11-25:
Your model works well under Tensorrt 6.x;
anyway, i will give it a try.

jkjung-avt, 2019-11-25:
I see,  Thanks for the update.

Ndron, 2020-01-30:
same problem

jkjung-avt, 2020-01-30:
@Ndron  Could you check and make sure you have patched "graphsurgeon/node_manipulation.py"?  More specifically, you have to add the following line of code into the file:

    +    node.attr["dtype"].type = 1

Reference:

- https://github.com/jkjung-avt/tensorrt_demos/blob/master/README_x86.md
- https://github.com/AastaNV/TRT_object_detection ("Update graphsurgeon converter")

Ndron, 2020-01-30:
@jkjung-avt Fixed , thanks

closed 2020-01-30 by jkjung-avt

aimuch, 2020-07-02:
> @jkjung-avt Fixed , thanks

Hi， same problem，how did you solved it? 
Thanks