jkjung-avt/tensorrt_demos #26: the video stream delays for 300ms

xzGithu, 2019-12-23:
hello, I run this code on my jetson nano for a live stream detect but the video has about 300ms delay, and the fps for me is about 10 with the trt_ssd.py.  Do you have there situations, Thanks. I use the jetpack4.2 and tensorrt5.0.6.3,

jkjung-avt, 2019-12-23:
1. If you are testing with an RTSP stream, there's a '--latency' (milliseconds) command-line option which you could use to adjust latency of the decoding pipeline.
2. Did you set Jetson Nano to 10W model? (sudo nvpmodel -m 0; sudo jetson_clocks)
3. What's the frame rate if you test with an input image (jpg)?

xzGithu, 2019-12-23:
Thanks for your reply.
1.Yes it is the RTSP stream and I saw the --latency args now and adjust it 
2. I used the 5W model on my jetson nano, and is it has more influence on the fps?
3. If i test a picture  with mobile-v1 the fps is about 15 less than yours, is there any thing to optimize for the nano
Thanks

jkjung-avt, 2019-12-23:
Power mode directly affects the "clock rate" Nano is running at.  All my measurements were done with Nano set to 10W mode.

FPS would definitely be lower if you set Nano to 5W mode.

xzGithu, 2019-12-23:
Yeah I got it, I will change it to 10w mode and test further ,Thanks for your help

closed 2019-12-24 by xzGithu