jkjung-avt/tensorrt_demos #31: Issue while trying with ssd_inception_v2_coco

SahilChachere, 2020-01-03:
while converting frozen_inference_graph.pb file to uff format, I downloaded from TensorFlow zoo I got key error:  'image_tensor', can you tell why i am getting this error. 

for reference: 

UFF Version 0.6.5
=== Automatically deduced input nodes ===
[name: "Input"
op: "Placeholder"
input: "image_tensor:0"
attr {
  key: "dtype"
  value {
    type: DT_FLOAT
  }
}
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: 1
      }
      dim {
        size: 3
      }
      dim {
        size: 300
      }
      dim {
        size: 300
      }
    }
  }
}
]

jkjung-avt, 2020-01-03:
Just remove that input by enabling this line of code: https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py#L160

SahilChachere, 2020-01-03:
> I think the issue could be fixed by just removing the "image_tensor:0" input by enabling this line of code: https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py#L160

I solved this but now at last step during conversion I am getting core dumped error.

for reference: 

UFF Output written to /home/sys-admin/Downloads/Projects/tensorrt_demos/ssd/tmp_inception_v2_coco.uff
UFF Text Output written to /home/sys-admin/Downloads/Projects/tensorrt_demos/ssd/tmp_inception_v2_coco.pbtxt
[TensorRT] INFO: Detected 1 inputs and 2 output network tensors.
python3: nmsPlugin.cpp:139: virtual void nvinfer1::plugin::DetectionOutput::configureWithFormat(const nvinfer1::Dims*, int, const nvinfer1::Dims*, int, nvinfer1::DataType, nvinfer1::PluginFormat, int): Assertion `numPriors * numLocClasses * 4 == inputDims[param.inputOrder[0]].d[0]' failed.
./build_engines.sh: line 5:  7298 Aborted                 (core dumped) python3 build_engine.py ${model}

jkjung-avt, 2020-01-03:
Check input order of the 3 input tensors of the NMS_TRT node (look at your 'tmp_inception_v2_coco.pbtxt').

Reference: https://devtalk.nvidia.com/default/topic/1050465/jetson-nano/how-to-write-config-py-for-converting-ssd-mobilenetv2-to-uff-format/post/5333033/#5333033

SahilChachere, 2020-01-04:
> Check input order of the 3 input tensors of the NMS_TRT node (look at your 'tmp_inception_v2_coco.pbtxt').
> 
> Reference: https://devtalk.nvidia.com/default/topic/1050465/jetson-nano/how-to-write-config-py-for-converting-ssd-mobilenetv2-to-uff-format/post/5333033/#5333033

@jkjung-avt as you mentioned to check 'tmp_inception_v2_coco.pbtxt' file, I have gone through it and compared parameters with build_engine.py file, all are similar, still, I am getting core dumped error, have you tried with ssd_inception model or have you such issue, what solution can you suggest specifically for ssd_inception_v2 model?

for reference: 
in build_engine.py file - 

},
    'ssd_inception_v2_coco': {
        'input_pb':   os.path.abspath(os.path.join(
                          DIR_NAME, 'ssd_inception_v2_coco.pb')),
        'tmp_uff':    os.path.abspath(os.path.join(
                          DIR_NAME, 'tmp_inception_v2_coco.uff')),
        'output_bin': os.path.abspath(os.path.join(
                          DIR_NAME, 'TRT_ssd_inception_v2_coco.bin')),
        'num_classes': 90,
        'min_size': 0.2,
        'max_size': 0.95,
        'input_order': [1, 0, 2],  # order of loc_data, conf_data, priorbox_data
    },

what else changes I must do in code??

SahilChachere, 2020-01-04:
@jkjung-avt thanks for replies, i solved all issues and successfully run.

closed 2020-01-04 by SahilChachere