jkjung-avt/tensorrt_demos #33: ssd_inception_v2_coco custom trained model with my own dataset.

SahilChachere, 2020-01-04:
I tried with ssd_inception_v2_coco custom trained model with my own dataset and the following error occurred: - 
No. nodes: 721
UFF Output written to /home/sys-admin/Downloads/Projects/tensorrt_demos/ssd/tmp_inception_v2_coco.uff
UFF Text Output written to /home/sys-admin/Downloads/Projects/tensorrt_demos/ssd/tmp_inception_v2_coco.pbtxt
[TensorRT] ERROR: UffParser: Validator error: Cast: Unsupported operation _Cast
[TensorRT] ERROR: Network must have at least one output
Traceback (most recent call last):
  File "build_engine.py", line 216, in <module>
    main()
  File "build_engine.py", line 210, in main
    buf = engine.serialize()
AttributeError: 'NoneType' object has no attribute 'serialize'

SahilChachere, 2020-01-04:
Solved it by replacing input: "image_tensor: 0" by input: "Cast".

closed 2020-01-04 by SahilChachere

maky-hnou, 2020-03-15:
@SahilChachere  could you please mention in which file did you make that change?

jkjung-avt, 2020-03-16:
You could reference NVIDIA's original code: [model_ssd_inception_v2_coco_2017_11_17.py](https://github.com/AastaNV/TRT_object_detection/blob/master/config/model_ssd_inception_v2_coco_2017_11_17.py)

The modification should go into [build_engine.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/build_engine.py) in this repository.

SahilChachere, 2020-03-16:
> Solved it by replacing input: "image_tensor: 0" by input: "Cast".

graph.collapse_namespaces(namespace_plugin_map)
    graph.remove(graph.graph_outputs, remove_exclusive_dependencies=False)
    graph.find_nodes_by_op("NMS_TRT")[0].input.remove("Input")
    #if model == 'ssd_inception_v2_coco':
        #graph.find_nodes_by_name("Input")[0].input.remove("image_tensor:0")
        #graph.find_nodes_by_name("Input")[0].input.remove("Cast")

try above changes in 'build_engine.py' file @maky-hnou