jkjung-avt/tensorrt_demos #34: ssd_inception_v2_coco(custom trained model)

SahilChachere, 2020-01-04:
I am trying to convert - ssd_inception_v2_coco(custom trained model) but during runtime its throeing following error.

I am using:-
tensorflow==1.14.0
jetson nano==Jetpack 4.3 [L4T 32.3.1]

for reference: 

No. nodes: 720
UFF Output written to /home/sys-admin/Downloads/Projects/tensorrt_demos/ssd/tmp_inception_v2_coco.uff
UFF Text Output written to /home/sys-admin/Downloads/Projects/tensorrt_demos/ssd/tmp_inception_v2_coco.pbtxt
[libprotobuf FATAL /externals/protobuf/aarch64/10.0/include/google/protobuf/repeated_field.h:1408] CHECK failed: (index) < (current_size_): 
Traceback (most recent call last):
  File "build_engine.py", line 219, in <module>
    main()
  File "build_engine.py", line 210, in main
    parser.parse(spec['tmp_uff'], network)
RuntimeError: CHECK failed: (index) < (current_size_):

jkjung-avt, 2020-01-04:
There are version dependencies among these things, so I wouldn't recommend you to use the latest 'tensorflow/models' code to train your own model.

* Object detection API (the code gets updated very frequently, and the latest code on GitHub probably only works with the latest version of tensorflow
* tensorflow
* UFF & TensorRT

Reference: https://github.com/jkjung-avt/hand-detection-tutorial/issues/24#issuecomment-570561659

SahilChachere, 2020-01-04:
> There are version dependencies among these things, so I wouldn't recommend you to use the latest 'tensorflow/models' code to train your own model.
> 
> * Object detection API (the code gets updated very frequently, and the latest code on GitHub probably only works with the latest version of tensorflow
> * tensorflow
> * UFF & TensorRT
> 
> Reference: [jkjung-avt/hand-detection-tutorial#24 (comment)](https://github.com/jkjung-avt/hand-detection-tutorial/issues/24#issuecomment-570561659)

I have trained my model in tensorflow==1.14.0 and in another machine, so on jetson nano also i have installed tf==1.14.0, if my tf version in training machine and nano is different will it work?

jkjung-avt, 2020-01-04:
> Reference: jkjung-avt/hand-detection-tutorial#24 (comment)

One very key thing is to use an older version ('6518c1c') of object detection API.  For example,

```
git clone https://github.com/tensorflow/models.git
cd models
git checkout 6518c1c
```

> I have trained my model in tensorflow==1.14.0 and in another machine, so on jetson nano also i have installed tf==1.14.0, if my tf version in training machine and nano is different will it work?

I've tested the following and it worked.  I cannot guarantee other combination would also work though.

* ssd_mobilenet_v1_egohands and ssd_mobilenet_v2_egohands trained with tensorflow-1.12.0
* optimize the 2 models with TensorRT 6 (JetPack-4.3) and tensorflow-1.15.0 on Jetson Nano
* run inference with TensorRT 6 on Jetson Nano

PythonImageDeveloper, 2020-01-04:
Hi @jkjung-avt 
1- Has the TensorRT 6 more speed than TensorRT 5?
2- You tested the models with jetpack 4.3 and TensorFlow-1.15.0 on jetson. For training you use Tensorflow-1.12.x?

jkjung-avt, 2020-01-04:
1. TensorRT 6 engines run slightly faster than TensorRT 5 (on the Jetson Nano I tested).
2. Yes.

jkjung-avt, 2020-01-05:
@SahilChachere, alternatively you could check out this post on NVIDIA developer forum: https://devtalk.nvidia.com/default/topic/1069027/tensorrt/parsing-gridanchor-op-_gridanchor_trt-protobuf-repeated_field-h-1408-check-failed-index-lt-current_size_-/post/5415537/#5415537

The author says the fix works for the latest tensorflow object detection API.

SahilChachere, 2020-01-06:
> @SahilChachere, alternatively you could check out this post on NVIDIA developer forum: [...]

@jkjung-avt, will I have to train model again with the latest TensorFlow object detection API? or are there any tweaks only for the already trained model in the latest TensorFlow object detection API?

jkjung-avt, 2020-01-06:
I don't have more information about the fix mentioned in that NVIDIA forum post.

On the other hand, if you plan to retrain your SSD model, I suggest you to use my [hand-detection-tutorial](https://github.com/jkjung-avt/hand-detection-tutorial) code.  Here's another reference which might help: https://github.com/jkjung-avt/hand-detection-tutorial/issues/24

SahilChachere, 2020-01-06:
> I don't have more information about the fix mentioned in that NVIDIA forum post.
> 
> On the other hand, if you plan to retrain your SSD model, I suggest you to use my [hand-detection-tutorial](https://github.com/jkjung-avt/hand-detection-tutorial) code. Here's another reference which might help: [jkjung-avt/hand-detection-tutorial#24](https://github.com/jkjung-avt/hand-detection-tutorial/issues/24)

@jkjung-avt yes, the only problem I am facing now is in last step of conversion, .uff and .pbtxt files are written but script stops in the last step of conversion i.e. .bin file.

SahilChachere, 2020-01-06:
> I don't have more information about the fix mentioned in that NVIDIA forum post.
>
> On the other hand, if you plan to retrain your SSD model, I suggest you to use my [...]

also, you have mentioned that you trained using tensorflow==1.12.0 but the problem persists with tf==1.12.0 that it is not compatible with Cuda 10.0 and I have installed Cuda 10.0. I will have to downgrade to Cuda version 9.0

jkjung-avt, 2020-01-06:
I build tensorflow-1.12.0 on my x86_PC by myself.  CUDA 10.0 works.

SahilChachere, 2020-01-09:
Thank you for all the help, I have resolved my issue by providing correct input_order = [1, 2, 0] but before that, I trained models using models@6518c1c version only, it doesn't support trained models using -  models repo which I was using, need resolve dependency issues.

closed 2020-01-09 by SahilChachere