jkjung-avt/tensorrt_demos #35: UUF Convertor

PythonImageDeveloper, 2020-01-04:
Hi
I have some question:
1- What's the UUF converter? why we use this converter before converting to the TensorRT engine?
As you know we can convert the Tensorflow forzen graph (.pb) model to TernsorRT engine model directory and no need for UFF convertor, with TensorRT API:
```
trt_graph = trt.create_inference_graph(
    input_graph_def,
    outputs,
    max_batch_size,
    max_workspace_size_bytes,
    minimum_segment_size, 
    precision_mode="FP16") 
```
My quesion is that what's advantage of converting .pb (Tensorflow) >> . bin (UFF) >> .pb (TensorRT) instead .pb (Tensorflow) >>  .pb (TensorRT)?

jkjung-avt, 2020-01-04:
Please read TensorRT documentation: https://docs.nvidia.com/deeplearning/sdk/tensorrt-developer-guide/index.html#api

Only Caffe, UFF and ONNX parsers are available in current releases of TensorRT.

jkjung-avt, 2020-01-30:
Regarding your question 3 above,

- "bindings" refer to the inputs and outputs of a TensorRT engine
- For each binding, we allocate 1 "host" memory block and 1 "cuda" memory block.  The 2 memory blocks are of the same size.
- The "host" memory blocks should be blocks of contiguous memory location belonging to the CPU.
- Lilewise, the "cuda" memory blocks belong to the GPU.
- To do inference, we first copy image data to the input "host" memory.  This makes sure the data is contiguous and in the correct order.  Code: https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/ssd.py#L99
- Then we ask GPU to copy the data from "host" to "cuda" memory.  Code: https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/ssd.py#L101
- Next, we "execute" the Tensor engine.
- Finally, we copy the outputs from "cuda" back to "host" memory.
- Note that CUDA driver would make sure the memcpy's and engine execution are done sequentially (no race condition).

closed 2020-01-30 by jkjung-avt