jkjung-avt/tensorrt_demos #39: jetson tx2 yolov3_to_onnx.py bug

Gaondong, 2020-01-11:
[image]
I run the script yolov3_to_onnx.py  on jetsontx2 with jetpack4.2.1 onnx=1.4.1 and it still show the above code without change.

jkjung-avt, 2020-01-11:
Did you run 'download_yolov3.sh' script as described in step 3 of [Demo #4](https://github.com/jkjung-avt/tensorrt_demos#yolov3)?

```
$ cd ${HOME}/project/tensorrt_demos/yolov3_onnx
$ ./download_yolov3.sh
```

Gaondong, 2020-01-11:
> Did you run 'download_yolov3.sh' script as described in step 3 of [Demo #4](https://github.com/jkjung-avt/tensorrt_demos#yolov3)?
> 
> ```
> $ cd ${HOME}/project/tensorrt_demos/yolov3_onnx
> $ ./download_yolov3.sh
> ```

Thanks for your reply,  I just wait for a long time and it runs normally.
I think this is because the Compute Capability of tx2‘s CPU is not enough

closed 2020-01-30 by jkjung-avt