jkjung-avt/tensorrt_demos #40: ssd_resnet50_v1_fpn_shared_box_predictor_640x640_coco

PythonImageDeveloper, 2020-01-12:
Hi,
Have you experience the converting the **ssd_resnet50_v1_fpn_shared_box_predictor_640x640_coco** to uff and bin files?

jkjung-avt, 2020-01-13:
No.  I haven't tried.  Maybe you could share your experience?

PythonImageDeveloper, 2020-01-13:
Hi
If you know about this below definitions, please explain about this:

>     PriorBox = gs.create_plugin_node(name="GridAnchor", op="GridAnchor_TRT",
>         minSize=0.2,
>         maxSize=0.95,
>         aspectRatios=[1.0, 2.0, 0.5, 3.0, 0.33],
>         variance=[0.1,0.1,0.2,0.2],
>         featureMapShapes=[19, 10, 5, 3, 2, 1], 
>         numLayers=6
>     )
>     NMS = gs.create_plugin_node(
>         name="NMS",
>         op="NMS_TRT",
>         shareLocation=1,
>         varianceEncodedInTarget=0,
>         backgroundLabelId=0,
>         confidenceThreshold=1e-8,
>         nmsThreshold=0.6,
>         topK=100,
>         keepTopK=100,
>         numClasses=91,
>         inputOrder=[1, 0, 2],
>         confSigmoid=1,
>         isNormalized=1,
>         scoreConverter="SIGMOID"
>     )

PythonImageDeveloper, 2020-01-13:
I want to convert ssd_resnet50_v1_fpn_shared_box_predictor_640x640_coco to uff file, I get this error:
`[TensorRT] ERROR: UffParser: Parser error: FeatureExtractor/resnet_v1_50/fpn/top_down/nearest_neighbor_upsampling/mul: Invalid scale mode, nbWeights: 4`

jkjung-avt, 2020-01-14:
It looks like this "FeatureExtractor/resnet_v1_50/fpn/top_down/nearest_neighbor_upsampling/mul" node is doing "elementwise product" of 2 tensors with broadcasting.

According to [this table](https://docs.nvidia.com/deeplearning/sdk/tensorrt-support-matrix/index.html#layers-matrix), [IElementWiseLayer](https://docs.nvidia.com/deeplearning/sdk/tensorrt-developer-guide/index.html#elementwise-layer) does support "broadcasting".  So I think you could try to dig deeper:

- Does the "FeatureExtractor/resnet_v1_50/fpn/top_down/nearest_neighbor_upsampling/mul" get converted to an Elementwise Product OP in TensorRT engine?
- If not, try to use graphsurgeon to replace that node with an Elementwise Product OP.

PythonImageDeveloper, 2020-01-14:
If it's doing "elementwise product" of 2 tensors with broadcasting, So the UFF library doesn't support this layers, right?
[https://docs.nvidia.com/deeplearning/sdk/tensorrt-api/python_api/uff/Operators.html#gather](UFF Operations)

jkjung-avt, 2020-01-14:
Interesting...

1. I don't find an UFF Operator (in the documentation link you've provided) which could perform elementwise product with broadcasting.  So what UFF operator was written in the uff file for this layer?  Was it a "binary" layer with "mul" operation?

As you have said, TensorRT probably only supports such "mul" where the 2 input tensors are of the same shape.

2. I'm not sure if upgrading to Tensor 6 would solve the problem.  But that's a possibility.

3. I was thinking about adding a couple of [Tile Layers](https://stackoverflow.com/questions/49937434/what-is-the-purpose-of-tiling-layer-in-caffe) to convert the [N,1,1,2,1,2,1] tensor to [N,20,1,20,1,256] (matching the other one).  But UFF doesn't seem to have such an operation.

I don't have an answer for you right now.  But as stated in my previous comment, TensorRT's Elementwise OP does support broadcasting.  So this seems to be a limitation of UFF......

PythonImageDeveloper, 2020-01-14:
1- [Binary mul operation](https://docs.nvidia.com/deeplearning/sdk/tensorrt-api/python_api/uff/Operators.html?highlight=binary#binary)

jkjung-avt, 2020-03-13:
I missed this question earlier.  The UFF (I believe is 0.6.5) in TensorRT 6.x was tested against tensorflow-1.14.0 (according to UFF's own log).  So tensorflow 1.14 or 1.15 is better.  That being said, I think tensorflow-1.12.2 would likely work OK, too.

mw3155, 2020-04-03:
Hey @PythonImageDeveloper ,
I also want to try to run a SSD model with FPN on Jetson.

Did you have any success?

PythonImageDeveloper, 2020-04-04:
Hi
No, I currently working on another project, If you can run the model on jetson, please let me know.

jkjung-avt, 2021-01-04:
I think I won't spend more time looking into this issue ("ssd_resnet50_v1_fpn_shared_box_predictor_640x640_coco")...

closed 2021-01-04 by jkjung-avt