jkjung-avt/tensorrt_demos #41: mtcnn optimization is input dependent

flyingmrwang, 2020-01-14:
Your optimization of mtcnn is specifically designed for 720p input image. While other size may be   scaled or padded to fit into 720p. The failure case may be an input of portrait rather than landscape. In this case, images are heavily squeezed and make it impossible to detect small faces. Is there any plan to make it input independent?

jkjung-avt, 2020-01-14:
You are right.  I've optimized MTCNN PNet based for 1280x720 image inputs, and it does not work very well for portrait inputs.  But I don't plan to update the code to better support other cases for now.  I think interested readers should be able to modify 'mtcnn/det1_relu.prototxt' and 'utils/mtcnn.py' by themselves.

In order to handle input images of multiple different input shapes, one might consider prebuilding TRT-PNet with a few different input shapes and select the best one at runtime.

flyingmrwang, 2020-01-15:
Actually, I am trying to find a face alignment model for efficient batch inference. Maybe mtcnn is not the best choice. Anyway, your demo helped me a lot.

closed 2020-01-15 by flyingmrwang

jkjung-avt, 2020-01-15:
Glad to hear that.  Good luck with your own project.