jkjung-avt/tensorrt_demos #42: Why config.py different from ssd_mobilenet_v1_coco and ssd_mobilenet_v2_coco?

zychen2016, 2020-01-14:
When compare config.py between ssd_mobilenet_v1_coco and ssd_mobilenet_v2_coco ,
There some different context
like NMS input order and V1`MultipleGridAnchorGenerator/Concatenate": concat_priorbox`、V2` "Concatenate": concat_priorbox,`.

But why should use this?

jkjung-avt, 2020-01-15:
Looking at:

1. Error/warning messages thrown out by the UFF converter,
2. The 'ssd/tmp_v?_coco.pbtxt' file

Find the layers/ops which are not supported or could not be converted by UFF converter and TensorRT builder.  Then try to replace those layers with either supported nodes/ops or plugins.

zychen2016, 2020-01-15:
I'm not familiar with C++ and CUDA, So maybe using tensorflow-tensorrt could be a consider solution?

jkjung-avt, 2020-01-15:
You could just use TensorRT's python API, as I have done (demo #3 and demo #4) in this repository.

zychen2016, 2020-01-15:
> You could just use TensorRT's python API, as I have done (demo #3 and demo #4) in this repository.

But Like demo 3 ssd, also need `libflattenconcat.so` plugin written by C++ API.

PythonImageDeveloper, 2020-01-15:
Hi @zychen2016 
You prefer don't use TensorRT's python API because as not exist **libflattenconcat.so** in this API?

zychen2016, 2020-01-15:
> Hi @zychen2016
> You prefer don't use TensorRT's python API because as not exist **libflattenconcat.so** in this API?

Yes. It must written by user.But i'm not familiar with C++ API and CUDA.
In my understanding about TensorRT develop user guide, Now many nodes/ops are not supported by TensorRT now. Even There are TensorRT's python API, But it seems that also need the plugins written by C++.
if my understanding is wrong,Please tell me.  
Thank you very much.

jkjung-avt, 2020-01-15:
Indeed, TensorRT plug-ins must be programmed with C++ now (for current released versions).

On the other hands, NVIDIA has open-sourced a lot of plug-ins: [TensorRT/plugin/](https://github.com/NVIDIA/TensorRT/tree/master/plugin).  So unless your are using some cutting-edge layers, most likely you don't need to program plug-ins by yourself.

zychen2016, 2020-01-15:
Thanks. I will have a try with other models like ssdlite-mbv2 and ssdlitmbv2-fpn.
Then try to fix out errors.

PythonImageDeveloper, 2020-01-17:
@zychen2016 
Can you solved the probelm?

closed 2020-02-11 by jkjung-avt