jkjung-avt/tensorrt_demos #44: ERROR: nvcc not found with tensorflow-1.15 and jetpack-4.3

PythonImageDeveloper, 2020-01-20:
Hi,
I used your wheel TensorFlow 1.15 file for install and correctly is install. but when I run **./install.sh** in ssd folder, I get this error:

ERROR: nvcc not found
** Patch 'graphsurgeon.py' in TensorRT
patching file /usr/lib/python3.6/dist-packages/graphsurgeon/node_manipulation.py

** Making symbolic link of libflattenconcat.so
** Installation done

jkjung-avt, 2020-01-20:
1. "ERROR: nvcc not found" -> You just need to add /usr/local/cuda/bin" to your $PATH.  Reference: https://github.com/jkjung-avt/jetson_nano/issues/8#issuecomment-576001408

2. The difference in FPS is more related to TensorRT 5 vs. 6.  When you do inference with the optimized TensorRT engines, tensorflow is not used at all.

PythonImageDeveloper, 2020-01-20:
1- I had added /usr/local/cuda/bin" to $PATH and didn't work again. I have to clean the if statement.

2- TensorRT 5 faster than TensorRT 6, I tested ssd models with both v5 , v6.

closed 2020-01-30 by jkjung-avt