jkjung-avt/tensorrt_demos #46: ERROR: ValueError: not enough values to unpack (expected 2, got 1)

iAnkeet, 2020-01-21:
I Have trained model with YOLO v3, I Have two files yolov3.weights and yolov3.cfg. 
When I try to run yolov3_to_onnx.py with those files it throws these errors:

Traceback (most recent call last):
  File "yolov3_to_onnx.py", line 833, in <module>
    main()
  File "yolov3_to_onnx.py", line 791, in main
    layer_configs = parser.parse_cfg_file(cfg_file_path)
  File "yolov3_to_onnx.py", line 92, in parse_cfg_file
    layer_dict, layer_name, remainder = self._next_layer(remainder)
  File "yolov3_to_onnx.py", line 133, in _next_layer
    layer_param_block, remainder = remainder.split('\n\n', 1)
ValueError: not enough values to unpack (expected 2, got 1)

jkjung-avt, 2020-01-21:
In your "yolov3.cfg", make sure:

1. There is at least 1 empty line between 2 layers.
2. There is 1 empty line at the end of the file.

p.s. The darknet cfg parser code was provided by NVIDIA.  It only accepts cfg in the format stated above.

jkjung-avt, 2020-01-30:
I assume this issue has been resolved.

closed 2020-01-30 by jkjung-avt