jkjung-avt/tensorrt_demos #47: yolov3_onnx, onnx_to_tensorrt.py <tensorrt.tensorrt.Builder object at 0x7f5d678a78>, 0 bug

omeralierdemir, 2020-01-22:
Hi 
I tried this steps.
`

$ cd ${HOME}/project/tensorrt_demos/yolov3_onnx
$ ./download_yolov3.sh
$ python3 yolov3_to_onnx.py --model yolov3-tiny-416
$ python3 onnx_to_tensorrt.py --model yolov3-tiny-416

`

 "python3 onnx_to_tensorrt.py --model yolov3-tiny-416"  in this step I get the error:

`
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 119, in <module>
    main()
  File "onnx_to_tensorrt.py", line 115, in main
    _ = build_engine(onnx_file_path, engine_file_path, args.verbose)
  File "onnx_to_tensorrt.py", line 69, in build_engine
    with trt.Builder(TRT_LOGGER) as builder, builder.create_network(EXPLICIT_BATCH) as network, trt.OnnxParser(network, TRT_LOGGER) as parser:
TypeError: create_network(): incompatible function arguments. The following argument types are supported:
    1. (self: tensorrt.tensorrt.Builder) -> tensorrt.tensorrt.INetworkDefinition

Invoked with: <tensorrt.tensorrt.Builder object at 0x7f5d678a78>, 0
`

and I encountered this output during this command. I don't know if there is a problem " python3 yolov3_to_onnx.py --model yolov3-tiny-416"

`Layer of type yolo not supported, skipping ONNX node generation.
Layer of type yolo not supported, skipping ONNX node generation.
graph yolov3-tiny-416 (
  %000_net[FLOAT, 1x3x416x416]
) initializers (
  %001_convolutional_bn_scale[FLOAT, 16]
  %001_convolutional_bn_bias[FLOAT, 16]
  %001_convolutional_bn_mean[FLOAT, 16]
  %001_convolutional_bn_var[FLOAT, 16]
  %001_convolutional_conv_weights[FLOAT, 16x3x3x3]
  %003_convolutional_bn_scale[FLOAT, 32]
  %003_convolutional_bn_bias[FLOAT, 32]
  %003_convolutional_bn_mean[FLOAT, 32]
  %003_convolutional_bn_var[FLOAT, 32]
  %003_convolutional_conv_weights[FLOAT, 32x16x3x3]
  %005_convolutional_bn_scale[FLOAT, 64]
  %005_convolutional_bn_bias[FLOAT, 64]
  %005_convolutional_bn_mean[FLOAT, 64]
  %005_convolutional_bn_var[FLOAT, 64]
  %005_convolutional_conv_weights[FLOAT, 64x32x3x3]
  %007_convolutional_bn_scale[FLOAT, 128]
  %007_convolutional_bn_bias[FLOAT, 128]
  %007_convolutional_bn_mean[FLOAT, 128]
  %007_convolutional_bn_var[FLOAT, 128]
  %007_convolutional_conv_weights[FLOAT, 128x64x3x3]
  %009_convolutional_bn_scale[FLOAT, 256]
  %009_convolutional_bn_bias[FLOAT, 256]
`

Can you help me pls. I couldn't convert my yolov3 model to onnx_model.
I plan to use this onnx model to convert it to TRT C++ API

Michael-Dub, 2020-01-22:
I had the same issue, fixed it by replacing 
builder.create_network(EXPLICIT_BATCH)
with:
builder.create_network()

.. in onnx_to_tensorrt.py. Detection works fine now. The EXPLICIT_BATCH argument is specific for TensorRT7, I think.
I also had the message for 'Layer of type yolo not supported', but it all seemed to work nonetheless.

jkjung-avt, 2020-01-22:
Sorry, this is my bad.  I modified the code to support TensorRT 7 (for x86_64) a couple of days ago.  I did not do enough testing to make sure the modification does not break TensorRT 5 & 6.  I just pushed a fix into the repository.

Thanks to Michael-Dub for quicking answering the question.

omeralierdemir, 2020-01-24:
thanks your help.  Michael-Dub advise worked. I have few quastions ? My system tensorRT 6.0 on jetson tx2 and I try to convert my onnx_model to tensorRT C++ API  but I couldn't.  Do you have any idea for this quastions? I tried this repo https://github.com/onnx/onnx-tensorrt  but tx2 not suporrted tensorRT 7.0. https://devtalk.nvidia.com/default/topic/1069634/how-to-upgrade-tensorrt-to-7-0-version/?offset=7

jkjung-avt, 2020-01-25:
@omeralierdemir  Would you like to inference (do object detection) with TensorRT C++ API, or to convert a YOLOv3 model to TensorRT engine with C++?  If it's the former case, you could use the TensorRT engine created by my python scripts, and:

- Reference my [TrtGooglenet code](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trtNet.h#L56) for how to feed an input image ('float32' array) and do inference with the TensorRT engine.
- Implement the [preprocessing](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/yolov3.py#L59) and [postprocessing](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/yolov3.py#L68) code in C++.

I think a C++ implementation could run significantly faster than python if done properly.

closed 2020-01-30 by jkjung-avt