jkjung-avt/tensorrt_demos #53: The problem of yolov3_onnx

brian208579, 2020-02-03:
Followed the Demo #4: YOLOv3
when i execute this command at step3
python3 yolov3_to_onnx.py --model yolov3-tiny-416
I got the error below figure
[image]
I already update my protobuf to a recent version
Do you have any suggestions ?

jkjung-avt, 2020-02-03:
Are you using onnx==1.4.1?

brian208579, 2020-02-03:
@jkjung-avt yes, i do .

jkjung-avt, 2020-02-03:
Reference: https://github.com/onnx/onnx/issues/2534

I was using protobuf-3.8.0, and did not see this problem.  Could you try the same version?

brian208579, 2020-02-10:
@jkjung-avt 
sorry , i'm late for reply .
I try the protobuf-3.8.0 version and using sudo pip3 install protobuf==3.8.0 then it slove my problem
appreciate

closed 2020-02-11 by jkjung-avt