jkjung-avt/tensorrt_demos #55: trt_outputs = [output.reshape(shape) for output, shape ValueError: cannot reshape array of size 20577 into shape (1,255,19,19)

chenyuqing, 2020-02-11:
(base) tim@tim-System-Product-Name:~/workspace/tensorrt_demos$ python trt_yolov3.py --model yolov3-608 --usb --vid 0
[TensorRT] INFO: Glob Size is 128825344 bytes.
[TensorRT] INFO: Added linear block of size 47316992
[TensorRT] INFO: Added linear block of size 23658496
[TensorRT] INFO: Added linear block of size 11829248
[TensorRT] INFO: Added linear block of size 2957312
[TensorRT] INFO: Added linear block of size 1478656
[TensorRT] INFO: Added linear block of size 739328
[TensorRT] INFO: Found Creator ResizeNearest
[TensorRT] INFO: Found Creator ResizeNearest
[TensorRT] INFO: Deserialize required 1140534 microseconds.
Traceback (most recent call last):
  File "trt_yolov3.py", line 96, in <module>
    main()
  File "trt_yolov3.py", line 88, in main
    loop_and_detect(cam, trt_yolov3, conf_th=0.3, vis=vis)
  File "trt_yolov3.py", line 56, in loop_and_detect
    boxes, confs, clss = trt_yolov3.detect(img, conf_th)
  File "/home/tim/workspace/tensorrt_demos/utils/yolov3.py", line 473, in detect
    in zip(trt_outputs, self.output_shapes)]
  File "/home/tim/workspace/tensorrt_demos/utils/yolov3.py", line 472, in <listcomp>
    trt_outputs = [output.reshape(shape) for output, shape
ValueError: cannot reshape array of size 20577 into shape (1,255,19,19)

jkjung-avt, 2020-02-11:
Please refer to this post: https://github.com/jkjung-avt/tensorrt_demos/issues/51#issuecomment-580983383

20577 / 19 / 19 = **57**
(**14** + 5) * 3 = 57

I think your custom YOLOv3 model is detecting 14 classes of objects. So you should modify "category_num" to 14, and 255 in "output_shapes" to 57.

closed 2020-02-21 by jkjung-avt

chenyuqing, 2020-03-26:
It works, Thaks a lot ! ~ @jkjung-avt

jkjung-avt, 2020-03-26:
You are welcome.

jkjung-avt, 2020-06-13:
I've added a "--category_num" command-line option to make it easier to adapt my TensorRT YOLOv3 code to custom trained models.  Please check out my blog post [TensorRT YOLOv3 For Custom Trained Models](https://jkjung-avt.github.io/trt-yolov3-custom/) for details.