jkjung-avt/tensorrt_demos #56: slow with batch size > 1

PythonImageDeveloper, 2020-02-11:
Hi,
I set max_batch_size = 3 and I want to speed up model with 3 input image as parallel instead serial,
But I converted the model with batch_size = 3 correctly, when I run trt_ssd.py, I achieve this results:
for **1 batch_size** : process time is **0.002** sec with 1080 TI,
for **3 batch_size** : process time is **0.006** sec with 1080 TI,
That means the system process as serial not parallel, why?

jkjung-avt, 2020-02-11:
Does your processing time include image preprocessing?  (Do you get a different result with trt_ssd_async.py?)

PythonImageDeveloper, 2020-02-11:
time is only for **trt_ssd.detect(img, 0.3)**.
I don't test with trt_ssd_async.py.

I do following steps:
1- change input size from (1,3,300,300) to (3,3,300,300)
2- change to **builder.max_batch_size = 3**
4- change self.context.execute_async(
            **batch_size=3,**
            bindings=self.bindings,
            stream_handle=self.stream.handle)

Notice that when I comment self.stream.synchronize() in the ssd.py, I get first few result with 0.002 sec and then the time is growing then reach to 0.06, and then the line **self.stream.synchronize()** remain uncomment, I get 0.06 for all result, why?

jkjung-avt, 2020-02-12:
Instead of timing the whole trt_ssd.detect() function, I think it makes more sense for you to only time the "cuda.memcpy_xxx"s, "context.execute_async" and "cuda.stream.synchronize" in that function.

By the way, **the "self.stream.synchronize" call cannot be commented out.**  Otherwise, you cannot be sure GPU has finished processing the image.

PythonImageDeveloper, 2020-02-19:
This is my OCR custom model when I use **batch_size = 1, I get 0.02** sec and when I use **batch_size= 10, I get 0.2 sec**, which means, this batch_size is done as serializing, not parallel, why?
 
> All Time:  0.2079782485961914
> Copy:  0.20590543746948242
> 75U72866
> All Time:  0.20735740661621094
> Copy:  0.2055220603942871
> 75U72866
> All Time:  0.20698094367980957
> Copy:  0.20568513870239258
> 75U72866
> All Time:  0.2071824073791504
> Copy:  0.1931319236755371
> 75U72866
> All Time:  0.19458580017089844
> Copy:  0.004555463790893555
> 75U72866
> All Time:  0.00600123405456543
> Copy:  0.4196054935455322
> 75U72866
> All Time:  0.4210531711578369
> Copy:  0.2052767276763916
> 75U72866
> All Time:  0.20674490928649902
> Copy:  0.20600008964538574

jkjung-avt, 2020-05-01:
Duplicated issue: https://github.com/jkjung-avt/tensorrt_demos/issues/106

closed 2020-05-01 by jkjung-avt