jkjung-avt/tensorrt_demos #57: Could not install ONNX on jetson nano

SaddamBInSyed, 2020-02-11:
when i execute the below command I am getting below error,

jkjung-avt, 2020-02-12:
The problem is shown in this line.  That means "cmake" fails to find "protoc".

`Protobuf compiler not found`

Please install "protoc" (e.g. run my [install_protobuf-3.8.0.sh](https://github.com/jkjung-avt/jetson_nano/blob/master/install_protobuf-3.8.0.sh) script) and try again.

SaddamBInSyed, 2020-02-12:
@jkjung-avt 
thank you.

it worked.

at the same time, I would like to ask 

1. Is there any nvidia pretrained custom vehicle detection tensorRT model available?

2. I come across deepstream SDK that has a custom vehicle detection model. So 
Can I use the same model with your trt_yolo.py code? (so that the inference time will be less.)

hope you understand my questions.

please advise.

jkjung-avt, 2020-02-12:
The pretrained yolov3 "coco" models should be capable of detecting vehicles.  You could check out [this class list](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/yolov3_classes.py#L9).

I'm not sure if the TensorRT models included in DeepStream SDK would work with my scripts (most likely not).  You'll have to study that by yourself, or seek help from NVIDIA...

SaddamBInSyed, 2020-02-12:
thanks for the reply.

also, What are the things I need to do if I want to include your  trt_yolo.py (yolov3) code into my application?

got little confused Since I am a beginner.

please advise.

jkjung-avt, 2020-02-12:
The code is open-sourced.  You could use it by following the included LICENSE.

closed 2020-02-21 by jkjung-avt

Ncnguyen-max, 2021-05-29:
Hi,
When I do in this step:
$ cd ${HOME}/project/tensorrt_demos/plugins
$ make
I have this error:
"computes: 53
NVCCFLAGS: -gencode arch=compute_53,code=[sm_53,compute_53]
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_53,code=[sm_53,compute_53] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
/bin/sh: 1: nvcc: not found
Makefile:37: recipe for target 'yolo_layer.o' failed
make: * [yolo_layer.o] Error 127"
Pls help me, thanks you so much.

moritz157, 2022-07-03:
For future reference: I had a similiar error, however the problem wasn't that "protoc" wasn't installed at all, but that it required a newer version. 

Checking the protoc version using `protoc --version` revealed I had 3.0.0 installed.
I [upgraded the version by building from source](https://stackoverflow.com/a/47414689/9177952) to 3.5.0 and that fixed it.