jkjung-avt/tensorrt_demos #58: How to inference own yolo model ?

brian208579, 2020-02-11:
@jkjung-avt 
I try your thoughts in this Disqus post , and put my custom yolo weight & cfg in /yolov3_onnx.
Then excute python3 yolov3_to_onnx.py --model yolov3-416
I got the error below the figure
[image]
Can you give me some advice ?
Thanks you a lot ...

jkjung-avt, 2020-02-12:
I think this is because your cfg and weights files don't match (number of weights in the output convolutional layers).

The original "yolov3-tiny-416.cfg" is for detecting 80 classes.  So its output convolutional layers (right before the "yolo" layers) have (80 + 5) * 3 = 255 filters.

Say, if your custom model is for detecting only 1 class, number of filters in those layers should be modified to (1 + 5) * 3 = 18.

So you could find all lines of:

`filters=255`

and replace them with:

`filters=18`

brian208579, 2020-02-12:
您好，我看您是台灣人，我想說用中文表達應該會比英文清楚一點 ( 抱歉 英文太爛 ... )
因為 python3 yolov3_to_onnx.py 這行指令只接受 yolov3-tiny-xxx ro yolov3-xxx 等名稱
故我將我自己訓練的 weight 跟 cfg 改名為 yolov3-tiny.weights 與  yolov3-tiny.cfg
並將 download_yolov3.sh 修改成
set -e
echo
echo "Creating YOLOv3-Tiny-288 and YOLOv3-Tiny-416 configs..."
cat yolov3-tiny.cfg | sed -e '8s/width=416/width=288/' | sed -e '9s/height=416/height=288/' > yolov3-tiny-288.cfg
echo >> yolov3-tiny-288.cfg
ln -sf yolov3-tiny.weights yolov3-tiny-288.weights
cp yolov3-tiny.cfg yolov3-tiny-416.cfg
echo >> yolov3-tiny-416.cfg
ln -sf yolov3-tiny.weights yolov3-tiny-416.weights
echo
echo "Done."
生成不同尺寸的 weight 跟 cfg 檔
並按照您blog的那則留言作相對應的修改 ( 我的 class 為 3 故將 255 修改成 24 )
完成以上步驟後執行 python3 yolov3_to_onnx.py --model yolov3-416
就跑出如上圖片的錯誤了
請問你覺得是否還有其他地方或是以上步驟出錯了呢 ?
謝謝

jkjung-avt, 2020-02-13:
你訓練的model是YOLOv3還是YOLOv3-Tiny?  如果是YOLOv3的話，請將cfg和weights檔名中的 "-tiny" 刪掉再重試一次。

brian208579, 2020-02-13:
您好，感謝您的回覆
我使用的是 yolov3-tiny

saisubramani, 2020-02-13:
Try this @brian208579 as @jkjung-avt  said
hi i founded the error ! the error was we want to change the output shapes which was given in
self.output_shapes = [(batch_size, 18, 13, 13), (batch_size, 18, 26, 26)]
to
self.output_shapes = [(batch_size, 72, 13, 13), (batch_size, 72, 26, 26)]
where 72 is the filter size which we used during training the last layer.
num_of_calsses = 19
(num_of_classes +5)*3=72

brian208579, 2020-02-13:
@saisubramani  thx for your suggestion
I already try ( my model class is 3 )

if 'tiny' in model:
            self.output_shapes = [(1, **255**, h // 32, w // 32),
                                  (1, **255**, h // 16, w // 16)]
        else:
            self.output_shapes = [(1, **255**, h // 32, w // 32),
                                  (1, **255**, h // 16, w // 16),
                                  (1, **255**, h //  8, w //  8)]

to

if 'tiny' in model:
            self.output_shapes = [(1, **24**, h // 32, w // 32),
                                  (1, **24**, h // 16, w // 16)]
        else:
            self.output_shapes = [(1, 24, h // 32, w // 32),
                                  (1, **24**, h // 16, w // 16),
                                  (1, **24**, h //  8, w //  8)]

but i got the same error ...

saisubramani, 2020-02-14:
Can you share your yolov3-tiny-416.cfg file? so that i can have better understand.

brian208579, 2020-02-14:
@saisubramani ok , appreciate

[net]
# Testing
batch=64
subdivisions=8
# Training
# batch=64
# subdivisions=2
width=608
height=608
channels=3
momentum=0.9
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.001
burn_in=1000
max_batches = 50000
policy=steps
steps=40000,45000
scales=.1,.1

[convolutional]
batch_normalize=1
filters=16
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=64
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=128
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=256
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=512
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=1

[convolutional]
batch_normalize=1
filters=1024
size=3
stride=1
pad=1
activation=leaky

###########

[convolutional]
batch_normalize=1
filters=256
size=1
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=512
size=3
stride=1
pad=1
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=24
activation=linear

[yolo]
mask = 3,4,5
anchors = 10,14,  23,27,  37,58,  81,82,  135,169,  344,319
classes=3
num=6
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1

[route]
layers = -4

[convolutional]
batch_normalize=1
filters=128
size=1
stride=1
pad=1
activation=leaky

[upsample]
stride=2

[route]
layers = -1, 8

[convolutional]
batch_normalize=1
filters=256
size=3
stride=1
pad=1
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=24
activation=linear

[yolo]
mask = 0,1,2
anchors = 10,14,  23,27,  37,58,  81,82,  135,169,  344,319
classes=3
num=6
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1

brian208579, 2020-02-14:
@saisubramani I already solved my problem , the reason is I didn't add spaces at the bottom of cfg , thanks for your help @jkjung-avt @saisubramani
I will coles the comment
Appreciate !!!

closed 2020-02-14 by brian208579

saisubramani, 2020-02-15:
> @saisubramani I already solved my problem , the reason is I didn't add spaces at the bottom of cfg , thanks for your help @jkjung-avt @saisubramani
> I will coles the comment
> Appreciate !!!

its good @brian208579