jkjung-avt/tensorrt_demos #59: Mtcnn accuracy problem

SaddamBInSyed, 2020-02-12:
Thanks for your work.

I have carefully followed your build instructions and when I run the TensorRT mtcnn face detector on my jetson nano, I am getting low accuracy as like below
code detecting my hand as a face (continuously for some seconds ).

please advise.

[image]

jkjung-avt, 2020-02-12:
I think my TensorRT MTCNN result should be very close to the original Caffe model.

I did not train the original Caffe MTCNN model by myself.  If you'd like to improve its recall/precision, please look into how to re-train or fine tune it.

julayelei, 2020-02-17:
Hello, Saddam. You can also change values of MINSIZE for faces detected. Sometimes, some objects "look" like a face. By default, it uses 40 minzise in utils/mtcnn. Try changing it to a higher minsize!

SaddamBInSyed, 2020-02-18:
@jkjung-avt 
thanks for your advice

seems it works fine when I change the minsize from 40 to 100 pixels.

thank you.

jkjung-avt, 2020-02-18:
Thanks to @julayelei

closed 2020-02-21 by jkjung-avt