jkjung-avt/tensorrt_demos #60: Reshape Error while running Inference

saisubramani, 2020-02-13:
Hi , i followed your code and i tried to covert for my model all went good. i used my .cfg file and .weights file, it given all .onnx and .trt file. But the thing is what i cant able to run inference its showing cant be reshape i dont know why? I changed the categories =19 as my number of classes in utils/yolov3.py file

` trt_outputs = [output.reshape(shape) for output, shape
Value-error: cannot reshape array of size 12168 into shape (1,255,13,13)`

please give a detail explanation about the error.

jkjung-avt, 2020-02-13:
Please refer to this: https://github.com/jkjung-avt/tensorrt_demos/issues/51

saisubramani, 2020-02-13:
@jkjung-avt thanks i had founded the issue
hi i founded the error ! the error was we want to change the output shapes which was given in
self.output_shapes = [(batch_size, 18, 13, 13), (batch_size, 18, 26, 26)]
to
self.output_shapes = [(batch_size, 72, 13, 13), (batch_size, 72, 26, 26)]
where 72 is the filter size which we used during training the last layer.
num_of_calsses = 19
(num_of_classes +5)*3=72

closed 2020-02-13 by saisubramani