jkjung-avt/tensorrt_demos #61: Does tensorrt 4x support converting yolov3 model to onnx?

omeralierdemir, 2020-02-18:
Hi,
Does tensorrt 4x support converting yolov3 model to onnx?

I have a jetson tx2 development board. My systems:
Ubuntu 16.04
TensorRT 4x
Cuda 9
CuDnn 7

I want to convert my special yolov3 model conveting to onnx model then convert to yolov3.engine.
But When I tried, I got some mistake.
Does tensorrt 4x support converting yolov3 model to onnx then yolov3.engine? and Which onnx version should I install with pip?

Thanks for your help..

jkjung-avt, 2020-02-18:
NVIDIA did not provide "python API" for TensorRT 4 on the Jetson platforms.  So you cannot use code in this repository to optimize your YOLOv3 model with TensorRT 4 on Jetson TX2.

omeralierdemir, 2020-02-18:
hmmm,
So is there any support for C ++ API? I want to convert my yolo.onnx model to C++ API.

Thanks for your help..

jkjung-avt, 2020-02-18:
Yes, there's C++ API.  Please refer to [official documentation](https://docs.nvidia.com/deeplearning/sdk/tensorrt-developer-guide/index.html#c_topics) and sample code.

closed 2020-02-21 by jkjung-avt