jkjung-avt/tensorrt_demos #62: more Facial landmark in mtcnn

SaddamBInSyed, 2020-02-19:
@jkjung-avt ,
Thanks for your work.

how to get 68 points facial landmark (as like dlib) using MTCNN.
currently, it's giving only 5 facial points.
Since I want to do head pose estimation I am need of the other facial points

please advise.

jkjung-avt, 2020-02-19:
MTCNN is designed and trained with only 5 facial landmarks.  If you want to do 68 points, you need to train your own model.

Reference: [official web site of MTCNN](https://kpzhang93.github.io/MTCNN_face_detection_alignment/)

[image]

SaddamBInSyed, 2020-02-19:
thanks for the information

SaddamBInSyed, 2020-02-20:
@jkjung-avt ,

how to extract/crop left_eye and right_eye image from mtcnn facial landmark.
I want to find face drowsiness and other stuff.

please advise.

jkjung-avt, 2020-02-20:
The landmarks output by MTCNN already include left and right eyes.  I think you could use those to extract image crops of the eyes and maybe train a classifier to do drowsiness detection.

closed 2020-02-26 by jkjung-avt