jkjung-avt/tensorrt_demos #63: PyCuda logic error on jetson nano with latest image

SaddamBInSyed, 2020-02-19:
I have carefully followed your instruction and I can able to run the python yolo/ssd  sample code.

But When I integrate the same code into my multi-threaded application I am getting the below error.

```
yolov3.py", line 282, in allocate_buffers

stream = cuda.Stream()
pycuda._driver.LogicError: explicit_context_dependent failed: invalid device context - no currently active context?

```
please advise

SaddamBInSyed, 2020-02-19:
@jkjung-avt 
Also I am facing the issue similar to below

https://devtalk.nvidia.com/default/topic/1056268/tensorrt/tensorrt-do_inference-error/

Is there any particular reason for not implementing the yolo_async sample code?

I see only ssd_async code in the example.

jkjung-avt, 2020-02-20:
A "trt_yolov3_async.py" is very easy to implement.  You could just copy my "trt_ssd_async.py" and modify a few lines.  I remember I did that, but saw very little improvement on FPS.  So I did not put the code into my repository.

Regarding the problem you mentioned in the first post, it's likely because you did not initialize pycuda driver properly.  Please read the documentation or reference my "trt_ssd_sync.py" code.

closed 2020-02-26 by jkjung-avt