jkjung-avt/tensorrt_demos #64: my finetuned model is much slower

xmuszq, 2020-02-25:
Hi JK,

I tried the model in your repo in Nano, the  ssd_mobilenet_v1_egohands, the fps is 20. Then I trained/fine tuned the model with my own data from the same model (from model zoon). After converting to TRT in nano, the fps is only 12. 
Do you have any ideas about what factors that will affect the speed?

jkjung-avt, 2020-02-25:
If your custom model is detecting more classes of objects, it would be slower.

Otherwise, I'm not able to make guesses with the limited information you've provided.

xmuszq, 2020-02-25:
Hi Jk,

Thank you for your reply! 
I trained in TF1.12.0 with the your suggested detection API version. My Jetpack 4.2.2 and tensorrt version 5.0.6.3.  
I have only 2 classes. 
What i have tried is to lower down the 
        topK=100,
        keepTopK=100,
to 20 and increase the 
        'min_size': 0.05,
to 0.2
the fps can go up to 20. 
But these settings will lower the performance.

jkjung-avt, 2020-02-26:
This suggested your model probably spends a lot of time doing "NMS".  My guess is that your model outputs too many candidate bounding boxes.  As a result, the "NMS" op needs to spend a lot time comparing many bounding boxes.

So try the following:

* train your model better, or
* try to use a higher confidence threshold for NMS.

closed 2020-02-26 by xmuszq