jkjung-avt/tensorrt_demos #65: UFF Error on Custom mobilenetv2 ssd

GotG, 2020-02-26:
Hello, do you have any suggestions for the following error please? At the moment I am trying various suggestions from the nvidia forum without success. 
Thank you!

[TensorRT] INFO: UFFParser: parsing MultipleGridAnchorGenerator
[libprotobuf FATAL /home/erisuser/p4sw/sw/gpgpu/MachineLearning/DIT/externals/protobuf/aarch64/10.0/include/google/protobuf/repeated_field.h:1408] CHECK failed: (index) < (current_size_):
Traceback (most recent call last):
  File "build_engine.py", line 220, in <module>
    main()
  File "build_engine.py", line 211, in main
    parser.parse(spec['tmp_uff'], network)
RuntimeError: CHECK failed: (index) < (current_size_):

jkjung-avt, 2020-02-26:
Have you tried this?

[https://devtalk.nvidia.com/default/topic/1069027/tensorrt/parsing-gridanchor-op-_gridanchor_trt-protobuf-repeated_field-h-1408-check-failed-index-lt-current_size_-/?offset=3#5415537](https://devtalk.nvidia.com/default/topic/1069027/tensorrt/parsing-gridanchor-op-_gridanchor_trt-protobuf-repeated_field-h-1408-check-failed-index-lt-current_size_-/?offset=3#5415537)

Otherwise, you could avoid the problem by re-training the model with an older snapshot of tensorflow object detection API。

https://github.com/jkjung-avt/tensorrt_demos/issues/54#issuecomment-583707548

GotG, 2020-02-26:
Thank you, the link solved the issue. Just to clarify the indentation since it is not clear in the forum:
if base_anchor_size is None:
   base_anchor_size = [1.0, 1.0]
### added by yourself
base_anchor_size = tf.constant(base_anchor_size, dtype=tf.float32)
(the added line is inline with if, not inside the if)

closed 2020-02-26 by GotG