jkjung-avt/tensorrt_demos #66: Trt_async - multiple models

skyler1253, 2020-03-06:
@jkjung-avt Thank you for putting together this demo - I think it's one of the best ones on the Nano so far.  Quick question - I'm trying to add another serialized model after the ssd  engine.  I've basically reproduced the TrtSSD class and instantiated it with a different serialized model.  

```
logger.info('TrtThread: loading the TRT SSD engine...')
self.cuda_ctx = cuda.Device(0).make_context()  # GPU 0
self.trt_ssd = TrtSSD(self.ssd_model, Config.INPUT_HW)

logger.info('TrtThread: loading the face Detection engine...')
# self.cuda_ctx = cuda.Device(0).make_context()  # GPU 0
self.trt_face = faceTRT(self.face_trt_model, (1, 3, 256, 256))
logger.info('TrtThread: start running...')

self.running = True
while self.running:

    frame = self.cam.read()
    boxes, confs, clss = self.trt_ssd.detect(frame, self.conf_th)

    inputs, origin_img, center, scale = PreProcess(frame, bboxs, scores)
    inputs = inputs[:,[2,1,0]].numpy()
    faces = self.trt_face.detect(inputs)
```

I'm able to load both the TrtSSD class and my new model at the same time, but when I feed the results of the SSD into my new model, I get the error:

terminate called without an active exception
Aborted (core dumped)

Not a whole lot of help.  I'm wondering how you would set up the Trt_async model to run two models one after the other? It seems like my memory footprint is also pretty large, and i'm wondering if the two models should share more resources/context?

jkjung-avt, 2020-03-06:
Could you simply the code to a single-threaded implementation to see if you still get the same problem?

jkjung-avt, 2020-05-01:
Any update on this?  Or could I close the issue?

closed 2020-05-12 by jkjung-avt