jkjung-avt/tensorrt_demos #67: Cuda Error in NCHWToNCHHW2: 33

chennuo0125, 2020-03-09:
@jkjung-avt ,  when I add some lines in trt_yolov3.py as follows : 
```
"""trt_yolov3.py
This script demonstrates how to do real-time object detection with
TensorRT optimized YOLOv3 engine.
"""

import sys
import time
import argparse

import cv2
import pycuda.autoinit  # This is needed for initializing CUDA driver

from utils.yolov3_classes import get_cls_dict
from utils.yolov3 import TrtYOLOv3
from utils.camera import add_camera_args, Camera
from utils.display import open_window, set_display, show_fps
from utils.visualization import BBoxVisualization

==import tensorflow as tf==

WINDOW_NAME = 'TrtYOLOv3Demo'

def parse_args():
    """Parse input arguments."""
    desc = ('Capture and display live camera video, while doing '
            'real-time object detection with TensorRT optimized '
            'YOLOv3 model on Jetson Nano')
    parser = argparse.ArgumentParser(description=desc)
    parser = add_camera_args(parser)
    parser.add_argument('--model', type=str, default='yolov3-416',
                        choices=['yolov3-288', 'yolov3-416', 'yolov3-608',
                                 'yolov3-tiny-288', 'yolov3-tiny-416'])
    args = parser.parse_args()
    return args

def loop_and_detect(cam, trt_yolov3, conf_th, vis):
    """Continuously capture images from camera and do object detection.
    # Arguments
      cam: the camera instance (video source).
      trt_yolov3: the TRT YOLOv3 object detector instance.
      conf_th: confidence/score threshold for object detection.
      vis: for visualization.
    """
   
    ==a = tf.constant([1.0, 2.0], name="a")==
    ==b = tf.constant([3.0, 4.0], name="b")==
    ==result = a + b==
    ==with tf.Session() as sess:==
        ==print(sess.run(result))==

    full_scrn = False
    fps = 0.0
    tic = time.time()
    while True:
        if cv2.getWindowProperty(WINDOW_NAME, 0) < 0:
            break
        img = cam.read()
        if img is not None:
            boxes, confs, clss = trt_yolov3.detect(img, conf_th)
            img = vis.draw_bboxes(img, boxes, confs, clss)
            img = show_fps(img, fps)
            cv2.imshow(WINDOW_NAME, img)
            toc = time.time()
            curr_fps = 1.0 / (toc - tic)
            # calculate an exponentially decaying average of fps number
            fps = curr_fps if fps == 0.0 else (fps*0.95 + curr_fps*0.05)
            tic = toc
        key = cv2.waitKey(1)
        if key == 27:  # ESC key: quit program
            break
        elif key == ord('F') or key == ord('f'):  # Toggle fullscreen
            full_scrn = not full_scrn
            set_display(WINDOW_NAME, full_scrn)

def main():
    args = parse_args()
    cam = Camera(args)
    cam.open()
    if not cam.is_opened:
        sys.exit('Failed to open camera!')

    cls_dict = get_cls_dict('coco')
    yolo_dim = int(args.model.split('-')[-1])  # 416 or 608
    trt_yolov3 = TrtYOLOv3(args.model, (yolo_dim, yolo_dim))

    cam.start()
    open_window(WINDOW_NAME, args.image_width, args.image_height,
                'Camera TensorRT YOLOv3 Demo')
    vis = BBoxVisualization(cls_dict)
    loop_and_detect(cam, trt_yolov3, conf_th=0.3, vis=vis)

    cam.stop()
    cam.release()
    cv2.destroyAllWindows()

if __name__ == '__main__':
    main()
```

jkjung-avt, 2020-03-09:
I think the problem is TensorRT failing to allocate GPU memory because tensorflow occupies all GPU memories.  You could avoid this problem by specifying "gpu_options.allow_growth = True" when creating the tensorflow session.

chennuo0125, 2020-03-09:
@jkjung-avt, thank your reply, i try your advise as follows : 
```
gpu_options = tf.GPUOptions(allow_growth = True)
with tf.Session(config=tf.ConfigProto(gpu_options=gpu_options)) as sess:
     print(sess.run(result))
```
but the error i mentioned is also exiting ...... :( :( :( , there is any other reson ?

jkjung-avt, 2020-03-09:
I tested the code (with "gpu_options") on my Jetson Nano, using "yolov3-416".  And it worked OK.

Please check your own environment (tensorflow installation, etc.).

chennuo0125, 2020-03-09:
expect for your good result, thank you very much

jkjung-avt, 2020-03-09:
The following works.  It seems that closing the tensorflow session would destroy the pycuda driver context (which was initialized by `import pycuda.autoinit`).  So we need to create a new context for the TensorRT engine.

```
def main():                                                                     
    args = parse_args()                                                         
    cam = Camera(args)                                                          
    cam.open()                                                                  
    if not cam.is_opened:                                                       
        sys.exit('Failed to open camera!')                                      
                                                                                
    import tensorflow as tf                                                     
    a = tf.constant([1.0, 2.0], name="a")                                       
    b = tf.constant([3.0, 4.0], name="b")                                       
    result = a + b                                                              
    with tf.Session() as sess:                                                  
        print(sess.run(result))                                                 
                                                                                
    import pycuda.driver as cuda                                                
    cuda_ctx = cuda.Device(0).make_context()  # GPU 0                           
                                                                                
    cls_dict = get_cls_dict('coco')                                             
    yolo_dim = int(args.model.split('-')[-1])  # 416 or 608                     
    trt_yolov3 = TrtYOLOv3(args.model, (yolo_dim, yolo_dim))                    
                                                                                
    cam.start()                                                                 
    open_window(WINDOW_NAME, args.image_width, args.image_height,               
                'Camera TensorRT YOLOv3 Demo')                                  
    vis = BBoxVisualization(cls_dict)                                           
    loop_and_detect(cam, trt_yolov3, conf_th=0.3, vis=vis)                      
                                                                                
    cuda_ctx.pop()                                                              
                                                                                
    cam.stop()                                                                  
    cam.release()                                                               
    cv2.destroyAllWindows()
```

chennuo0125, 2020-03-09:
it's mean that can't use tf.Session() in loop_and_dectect() interface ?  i get reply from nvidia who said that tensorrt and tensorflow can be used together, but no information about how to use and attentions :( :( :(

chennuo0125, 2020-03-09:
[https://devtalk.nvidia.com/default/topic/1072901/cuda-error-in-nchwtonchhw2-33/?offset=5#5435705](url)

jkjung-avt, 2020-03-09:
Yes.  You should be able to use TensorRT and TensorFlow in the same program/process.  You just need to make sure they are executing under their own cuda execution context and don't mess with each other.

chennuo0125, 2020-03-09:
i get ! thank you very much for your attention to solve my problem, i will excute the code in same frame  (tensorrt or tensorflow)

chennuo0125, 2020-03-10:
i dont get any useful information about how to use tensorrt and tensorflow together properly :( , you said that make sure tensorflow and tensorrt execute under their own cuda execution context, do you have any idea to run tf.Session() in loop_and_detect() interface ? thank you.

jkjung-avt, 2020-03-10:
You could reference the code snippet in https://github.com/jkjung-avt/tensorrt_demos/issues/67#issuecomment-596398113.

Otherwise, you might need to read [pycuda documentation](https://documen.tician.de/pycuda/) and undestand how to manage cuda contexts.

chennuo0125, 2020-03-10:
ok, thank you for your reply :)

closed 2020-03-13 by jkjung-avt

gzchenjiajun, 2020-03-26:
Hi, I have a question now, which is "Cuda Error in nchwtonchhw2:33".
"Cuda Error in NCHWTONCHHW2: 33 (invalid resource handle) ", How to solve it? - Jetson & Embedded Systems/Jetson Nano - NVIDIA Developer associations
https://forums.developer.nvidia.com/t/cuda-error-in-nchwtonchhw2-33-invalid-resource-handle-how-to-solve-it/113137/8

I haven't solved this problem for two weeks. I need your help urgently. Thank you.

@jkjung-avt

gzchenjiajun, 2020-03-26:
@chennuo0125 Have you solved it? I introduced a trace model (pbfile) and TRT, and reported the same error as you

jkjung-avt, 2020-03-27:
The following example shows how you could manually set cuda context to make it work.  I'm not sure if this is the best solution, though.  You might want to study how to do this more efficiently.

```
"""trt_yolov3.py

This script demonstrates how to do real-time object detection with
TensorRT optimized YOLOv3 engine.
"""

import sys
import time
import argparse

import cv2
import tensorflow as tf
import pycuda.driver as cuda
import pycuda.autoinit  # This is needed for initializing CUDA driver

from utils.yolov3_classes import get_cls_dict
from utils.yolov3 import TrtYOLOv3
from utils.camera import add_camera_args, Camera
from utils.display import open_window, set_display, show_fps
from utils.visualization import BBoxVisualization

WINDOW_NAME = 'TrtYOLOv3Demo'

def parse_args():
    """Parse input arguments."""
    desc = ('Capture and display live camera video, while doing '
            'real-time object detection with TensorRT optimized '
            'YOLOv3 model on Jetson Nano')
    parser = argparse.ArgumentParser(description=desc)
    parser = add_camera_args(parser)
    parser.add_argument('--model', type=str, default='yolov3-416',
                        choices=['yolov3-288', 'yolov3-416', 'yolov3-608',
                                 'yolov3-tiny-288', 'yolov3-tiny-416'])
    args = parser.parse_args()
    return args

def loop_and_detect(cam, trt_yolov3, cuda_ctx, conf_th, vis):
    """Continuously capture images from camera and do object detection.

    # Arguments
      cam: the camera instance (video source).
      trt_yolov3: the TRT YOLOv3 object detector instance.
      cuda_ctx: cuda context for trt_yolov3
      conf_th: confidence/score threshold for object detection.
      vis: for visualization.
    """

    counter = tf.Variable(0., dtype=tf.float32)
    init = tf.global_variables_initializer()

    config = tf.ConfigProto(gpu_options=tf.GPUOptions(allow_growth=True))
    sess = tf.Session(config=config)
    sess.run(init)

    full_scrn = False
    fps = 0.0
    tic = time.time()
    while True:
        if cv2.getWindowProperty(WINDOW_NAME, 0) < 0:
            break
        img = cam.read()
        counter = counter + 1.
        print(sess.run(counter))
        if img is not None:
            cuda_ctx.push()  # set this as the active context
            boxes, confs, clss = trt_yolov3.detect(img, conf_th)
            img = vis.draw_bboxes(img, boxes, confs, clss)
            img = show_fps(img, fps)
            cv2.imshow(WINDOW_NAME, img)
            toc = time.time()
            curr_fps = 1.0 / (toc - tic)
            # calculate an exponentially decaying average of fps number
            fps = curr_fps if fps == 0.0 else (fps*0.95 + curr_fps*0.05)
            tic = toc
        key = cv2.waitKey(1)
        if key == 27:  # ESC key: quit program
            break
        elif key == ord('F') or key == ord('f'):  # Toggle fullscreen
            full_scrn = not full_scrn
            set_display(WINDOW_NAME, full_scrn)

    cuda_ctx.detach()
    sess.close()

def main():
    args = parse_args()
    cam = Camera(args)
    cam.open()
    if not cam.is_opened:
        sys.exit('Failed to open camera!')

    cuda.init()
    cuda_ctx = cuda.Device(0).make_context()  # GPU 0

    cls_dict = get_cls_dict('coco')
    yolo_dim = int(args.model.split('-')[-1])  # 416 or 608
    trt_yolov3 = TrtYOLOv3(args.model, (yolo_dim, yolo_dim))

    cam.start()
    open_window(WINDOW_NAME, args.image_width, args.image_height,
                'Camera TensorRT YOLOv3 Demo')
    vis = BBoxVisualization(cls_dict)
    loop_and_detect(cam, trt_yolov3, cuda_ctx, conf_th=0.3, vis=vis)

    cam.stop()
    cam.release()
    cv2.destroyAllWindows()

if __name__ == '__main__':
    main()
```

jkjung-avt, 2020-08-16:
Refer to: https://stackoverflow.com/a/59484474/7596504

jinfagang, 2022-01-05:
@jkjung-avt Hi, i have 2 tensorrt models runs at the same time and got this issue as well. Does there any way to do it? In my understanding, both tensorrt models would run just under the same cuda context?

But I got:

```
[01/05/2022-18:26:58] [TRT] [E] 1: [shapeHostToDeviceRunner.cpp::execute::53] Error Code 1: Cuda Runtime (invalid resource handle)
(1, 3, 416, 256)
[01/05/2022-18:26:58] [TRT] [E] 1: [reformat.cu::NCHWToNHWC::1436] Error Code 1: Cuda Runtime (invalid resource handle)
```

jkjung-avt, 2022-01-06:
@jinfagang  If you don't use multiple threads, I don't think you would encounter any problem.

So I guess you do use threads in your code.  Then the key thing to do is to push/pop the default cuda context before/after you do inference with the TensorRT engine context.

Refer to the following code snippet for details.

1. https://github.com/jkjung-avt/tensorrt_demos/issues/276#issuecomment-720275167
2. https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/utils/yolo_with_plugins.py#L280
3. https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/utils/yolo_with_plugins.py#L315-L316
4. https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/utils/yolo_with_plugins.py#L323-L324

jinfagang, 2022-01-06:
@jkjung-avt  thanks. I got it runs now. But when  I try destory the cuda_ctx, I got some repeat release error:

```
-------------------------------------------------------------------
PyCUDA ERROR: The context stack was not empty upon module cleanup.
-------------------------------------------------------------------
A context was still active when the context stack was being
cleaned up. At this point in our execution, CUDA may already
have been deinitialized, so there is no way we can finish
cleanly. The program will be aborted now.
```

I have using a wrapper class for trt inference, and I will call it twice for 2 different models. Does there a way to handle these cuda_ctx release?

I just make_context outside of classes, and in both class, I have a deconstruct function like:

```
   def __del__(self):
        """Free CUDA memories"""
        print('start deconstructing..')
        del self.engine
        del self.context
        del self.stream
        del self.outputs
        del self.inputs
        print('gpu resources released.')
```

i think the error comes from conflict between __del__ and cuda_ctx del

jkjung-avt, 2022-01-06:
If you are using the default context created by pycuda (`pycuda.autoinit.context`), you probably don't need to call `del` on it by yourself.

jinfagang, 2022-01-06:
@jkjung-avt  am using a manually maked context, and inside wrapper, it will call pop and push as Nvidia guided way for mutiple models (if not it will raise error like this ops issue).

```
cuda_ctx = cuda.Device(0).make_context()
    detector = Detector(engine_det_f, cuda_ctx=cuda_ctx, show=args.show)
    keyp2d = TensorRTInferencer(engine_pose_f, cuda_ctx=cuda_ctx)
```

Does it need call __del__ above inside wrapper? and how to destory cuda_ctx properly in this situation?

jkjung-avt, 2022-01-06:
Please refer to my code example in "trt_ssd_async.py".

https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/trt_ssd_async.py#L111

jinfagang, 2022-01-06:
@jkjung-avt the question is error raised:

```
-------------------------------------------------------------------
PyCUDA ERROR: The context stack was not empty upon module cleanup.
-------------------------------------------------------------------
A context was still active when the context stack was being
cleaned up. At this point in our execution, CUDA may already
have been deinitialized, so there is no way we can finish
cleanly. The program will be aborted now.
Use Context.pop() to avoid this problem.
```

jkjung-avt, 2022-01-06:
That probably means you did more `pop()`s than `push()`s.  You should do exactly 1 `pop()` after each `push()`.  Please review your code.