jkjung-avt/tensorrt_demos #68: Cask Error (Cask Convoluto

bikramb98, 2020-03-10:


jkjung-avt, 2020-03-10:
You are going in the right direction.  You just need to figure out how Flask creates threads.  And make sure you "make cuda context", "create tensorrt execution context" and "do inferencing" all in the same thread.

jkjung-avt, 2020-04-15:
Closing this issue since there's no update for over a month.

closed 2020-04-15 by jkjung-avt

jkjung-avt, 2020-09-14:
Please find a solution here: https://github.com/jkjung-avt/tensorrt_demos/issues/213#issuecomment-691826122.  Be sure to `git pull` the latest code from this repo.

In addition, @Papageno2 is kind enough to share full working source code here: https://github.com/jkjung-avt/tensorrt_demos/issues/213#issuecomment-691940559.  I think it would help.