jkjung-avt/tensorrt_demos #69: Error with yolo-608 but not 416

Kmarconi, 2020-03-11:
Hi jkjung-avt ! I just tried the trt_yolov3 script with my yolov3-416 engine and I'm around 17 FPS on my Xavier. I wanted to test the same script with a 608 engine but I got this error : 

`[cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument
`
Is it a known issue ? 

Regards :)

jkjung-avt, 2020-03-16:
I have not encountered this problem before.

Maybe you could use the debugger (pdb) and check whether sizes of "inp.host" (host memory buffer) and "inp.device" (device/GPU memory buffer) are the same.

jkjung-avt, 2020-04-15:
@Kmarconi, Any update on this issue?

Kmarconi, 2020-04-15:
Hi, thanks for reminding me this post, I do have found any ways to solve it for the moment, will keep you updated

closed 2020-05-27 by jkjung-avt

Onicc, 2020-07-04:
Hi,How to solve it,I have the same problem,Thank you!

Kmarconi, 2020-07-06:
I think you should check what @jkjung-avt said here : 

> I have not encountered this problem before.
> 
> Maybe you could use the debugger (pdb) and check whether sizes of "inp.host" (host memory buffer) and "inp.device" (device/GPU memory buffer) are the same.