jkjung-avt/tensorrt_demos #70: onnx.onnx_cpp2py_export.checker.ValidationError: Op registered for Upsample is deprecated in domain_version of 11

chennuo0125, 2020-03-19:
hello @jkjung-avt , when i using follow commandline : 
```
python3 yolov3_to_onnx.py --model yolov3-288
```
and occur follows error : 
```
  %104_convolutional_lrelu = LeakyRelu[alpha = 0.1](%104_convolutional_bn)
  %105_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [3, 3], strides = [1, 1]](%104_convolutional_lrelu, %105_convolutional_conv_weights)
  %105_convolutional_bn = BatchNormalization[epsilon = 1e-05, momentum = 0.99](%105_convolutional, %105_convolutional_bn_scale, %105_convolutional_bn_bias, %105_convolutional_bn_mean, %105_convolutional_bn_var)
  %105_convolutional_lrelu = LeakyRelu[alpha = 0.1](%105_convolutional_bn)
  %106_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [1, 1], strides = [1, 1]](%105_convolutional_lrelu, %106_convolutional_conv_weights, %106_convolutional_conv_bias)
  return %082_convolutional, %094_convolutional, %106_convolutional
}
Traceback (most recent call last):
  File "yolov3_to_onnx.py", line 819, in <module>
    main()
  File "yolov3_to_onnx.py", line 812, in main
    onnx.checker.check_model(yolov3_model_def)
  File "/usr/local/lib/python3.6/dist-packages/onnx/checker.py", line 91, in check_model
    C.check_model(model.SerializeToString())
onnx.onnx_cpp2py_export.checker.ValidationError: Op registered for Upsample is deprecated in domain_version of 11

==> Context: Bad node spec: input: "085_convolutional_lrelu" input: "086_upsample_scale" output: "086_upsample" name: "086_upsample" op_type: "Upsample" attribute { name: "mode" s: "nearest" type: STRING }

```
how to solve this problem ? expect for your advise, thank you

jkjung-avt, 2020-03-19:
Are you using onnx version **1.4.1**?  If not, I can't guarantee the code would work without modifications.

chennuo0125, 2020-03-19:
@jkjung-avt thank you, my onnx is 1.5.0, i swith to 1.4.1 version and the result is ok :) :) :)

closed 2020-03-19 by chennuo0125