jkjung-avt/tensorrt_demos #71: ValueError: cannot reshape array of size 1379040 into shape (1,255,13,13)

chennuo0125, 2020-03-23:
hello @jkjung-avt, when i generate int8-yolov3-416.trt by code : [https://github.com/chennuo0125/yolov3_int8_converter](url)  
and then execute : 
move int8-yolov3-416.trt yolov3-416.trt
python3 trt_yolov3.py --model yolov3-416
but get follow error : 
```
Traceback (most recent call last):
  File "trt_yolov3.py", line 97, in <module>
    main()
  File "trt_yolov3.py", line 91, in main
    loop_and_detect(trt_yolov3, conf_th=0.3, vis=vis)
  File "trt_yolov3.py", line 64, in loop_and_detect
    boxes, confs, clss = trt_yolov3.detect(img, conf_th)
  File "/home/xavier/python_ws/tensorrt_sort/utils/yolov3.py", line 473, in detect
    in zip(trt_outputs, self.output_shapes)]
  File "/home/xavier/python_ws/tensorrt_sort/utils/yolov3.py", line 472, in <listcomp>
    trt_outputs = [output.reshape(shape) for output, shape
ValueError: cannot reshape array of size 1379040 into shape (1,255,13,13)
```
you know why this happend ? expect for your answer

chennuo0125, 2020-03-23:
my platform : jetson xavier
os : jetpack 4.3

chennuo0125, 2020-03-24:
follow this repo [https://github.com/mynotwo/yolov3_tensorRT_int8_calibration](url) to generate int8 model can solve the problem, my pre provided repo has problem

closed 2020-03-24 by chennuo0125

jkjung-avt, 2020-03-24:
Great to hear that you have the problem solved.