jkjung-avt/tensorrt_demos #73: eval_yolov3.py error

QZ-cmd, 2020-03-27:
Hello, I got this error when testing the map of yolov3
TypeError: 'numpy.float64' object cannot be interpreted as an integer
Is the version of numpy wrong? My version is 1.18.1
looking forward to your reply thanks

jkjung-avt, 2020-03-27:
Please downgrade "onnx" to version 1.4.1.

QZ-cmd, 2020-03-27:
Thank you for your prompt reply, my onnx is 1.4.1 version; pycuda2019.1.2;
tensorRT6.0.1.10
It seems that something is wrong with the numpy type. I don't know exactly what went wrong. I look forward to your reply.

ntu@611:~/t2tproject/tensorrt_demos$ python3 eval_yolov3.py --model yolov3-416
100% (5000 of 5000) |###############################################################| Elapsed Time: 0:28:31 Time:  0:28:31
loading annotations into memory...
Done (t=2.07s)
creating index...
index created!
Loading and preparing results...
DONE (t=3.91s)
creating index...
index created!
Traceback (most recent call last):
  File "/home/ubuntu/.local/lib/python3.6/site-packages/numpy/core/function_base.py", line 117, in linspace
    num = operator.index(num)
TypeError: 'numpy.float64' object cannot be interpreted as an integer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "eval_yolov3.py", line 95, in <module>
    main()
  File "eval_yolov3.py", line 87, in main
    cocoEval = COCOeval(cocoGt, cocoDt, 'bbox')
  File "/home/ubuntu/.local/lib/python3.6/site-packages/pycocotools/cocoeval.py", line 76, in __init__
    self.params = Params(iouType=iouType) # parameters
  File "/home/ubuntu/.local/lib/python3.6/site-packages/pycocotools/cocoeval.py", line 527, in __init__
    self.setDetParams()
  File "/home/ubuntu/.local/lib/python3.6/site-packages/pycocotools/cocoeval.py", line 507, in setDetParams
    self.iouThrs = np.linspace(.5, 0.95, np.round((0.95 - .5) / .05) + 1, endpoint=True)
  File "<__array_function__ internals>", line 6, in linspace
  File "/home/ubuntu/.local/lib/python3.6/site-packages/numpy/core/function_base.py", line 121, in linspace
    .format(type(num)))
TypeError: object of type <class 'numpy.float64'> cannot be safely interpreted as an integer.

jkjung-avt, 2020-03-27:
If you've installed "pycocotools" from GitHub repository, please remove it and use pip installed version instead.

```
$ sudo pip3 uninstall pycocotools
$ sudo pip3 install pycocotools
```

QZ-cmd, 2020-03-28:
Thank you for your patience, I followed the steps to install pycocotools 2.0.0 with pip
progressbar2
3.50.1
After retesting the map, it is still the same numpy error. Hope to get your reply again, thank you very much

jkjung-avt, 2020-03-28:
The problem is in "pycocotools" code.  The 3rd argument to [numpy.linspace()](https://docs.scipy.org/doc/numpy/reference/generated/numpy.linspace.html) should be an "int".  The code (as follows), that throws exceptions, uses `np.round((0.95 - .5) / .05)` in the 3rd argument.  That's why python throws a `TypeError: object of type <class 'numpy.float64'> cannot be safely interpreted as an integer.`

```
self.iouThrs = np.linspace(.5, 0.95, np.round((0.95 - .5) / .05) + 1, endpoint=True)
```

This problem has been fixed in the latest "pycocotools" code.  You could refer to lines 506, 507, 517 and 518 of "cocoeval.py" here: https://github.com/cocodataset/cocoapi/blob/master/PythonAPI/pycocotools/cocoeval.py#L506

If `pip3 install pycocotools` does not work for you, you could manually patch those 4 lines of code to fix the problem.

QZ-cmd, 2020-03-28:
Thank you very much for your reply. The problem has been solved with your help

closed 2020-03-28 by jkjung-avt