jkjung-avt/tensorrt_demos #74: Use it to speed up your own model

YONGSONGWAN, 2020-03-27:
Hello, I would like to use your method to speed up my training of YOLOv3-tiny, only detector car and oerson .But when I finished training and generate the TRT file and performed the test, it prompted me ,
Traceback (most recent call last):
  File "trt_yolov3.py", line 97, in <module>
    main()
  File "trt_yolov3.py", line 89, in main
    loop_and_detect(cam, trt_yolov3, conf_th=0.3, vis=vis)
  File "trt_yolov3.py", line 57, in loop_and_detect
    boxes, confs, clss = trt_yolov3.detect(img, conf_th)
  File "/home/nano/my project/tensorrt_demos_1/utils/yolov3.py", line 473, in detect
    in zip(trt_outputs, self.output_shapes)]
  File "/home/nano/my project/tensorrt_demos_1/utils/yolov3.py", line 472, in <listcomp>
    trt_outputs = [output.reshape(shape) for output, shape
ValueError: cannot reshape array of size 3549 into shape (1,255,13,13)
what should i do？thanks!

jkjung-avt, 2020-03-28:
Please refer to this post: https://github.com/jkjung-avt/tensorrt_demos/issues/55

YONGSONGWAN, 2020-03-28:
> 请参阅此帖子：[＃55](https://github.com/jkjung-avt/tensorrt_demos/issues/55)

it works,thanks a lot!

closed 2020-03-28 by YONGSONGWAN