jkjung-avt/tensorrt_demos #75: Custom model based yolov3 error

GuyChriqui, 2020-03-31:
Hello,
First thanks for your hard work!

i saw your comment :

> Jiun-Kuei Jung Mod  Rafiuddin Khan • 3 months ago
> If you keep the same design of yolov3 (including anchor boxes, etc.), I think the required modification should be straightforward:
> 
> 1. Modify 'category_num' in 'PostprocessYOLO'. Code: https://github.com/jkjung-a...
> 2. Modify 'self.output_shapes'. For COCO, (80 + 5) * 3 = 255. For your own model, you need to replace 80 with your own number of classes. Code: https://github.com/jkjung-a...
> 3. Modify 'output_tensor_dims' in 'yolov3_to_onnx.py'. Same as above. You should replace 255 with the proper number for your model. Code: https://github.com/jkjung-a...

i managed to convert my own custom object detection from yolo to trt.
but when i run trt_yolov3.py i get the the following error:

> 
> [TensorRT] WARNING: Current optimization profile is: 0. Please ensure there are no enqueued operations pending in this context prior to switching profiles
> self.output_shapes [(1, 18, 13, 13), (1, 18, 26, 26), (1, 18, 52, 52)]
> Traceback (most recent call last):
> File "trt_yolov3.py", line 95, in <module>
> main()
> File "trt_yolov3.py", line 87, in main
> loop_and_detect(cam, trt_yolov3, conf_th=0.3, vis=vis)
> File "trt_yolov3.py", line 55, in loop_and_detect
> boxes, confs, clss = trt_yolov3.detect(img, conf_th)
> File "/home/guy/Documents/Gits/Github/tensorrt_demos/utils/yolov3.py", line 474, in detect
> in zip(trt_outputs, self.output_shapes)]
> File "/home/guy/Documents/Gits/Github/tensorrt_demos/utils/yolov3.py", line 473, in <listcomp>
> trt_outputs = [output.reshape(shape) for output, shape
> ValueError: cannot reshape array of size 194688 into shape (1,18,52,52)

ill be happy for some help.
Thanks!

jkjung-avt, 2020-04-01:
194688 = 1x18x104x104 = 1x72x52x52

I guess you have either increased the input image dimension, or increased the number of anchor boxes.

GuyChriqui, 2020-04-01:
Thank for the fast reply!
I didn't change any of them:
> input image dimension = 416X416
> anchors = 10,13,  16,30,  33,23,  30,61,  62,45,  59,119,  116,90,  156,198,  373,326

i tried to make it on another model i trained, and the same problem accord:
> ValueError: cannot reshape array of size 1525056 into shape (1,141,52,52)
(this model have 42 classes)
I trained the model based on 'darknet53.conv.74'
(as [AlexAB](https://raw.githubusercontent.com/AlexeyAB/darknet/master/README.md) in line 468 suggested: "for yolov3.cfg, yolov3-spp.cfg (154 MB): darknet53.conv.74")

This is my model config file :
> [net]
> # Testing
> # batch=1
> # subdivisions=1
> # Training
> batch=64
> subdivisions=32
> width=416
> height=416
> channels=3
> momentum=0.9
> decay=0.0005
> angle=0
> saturation = 1.5
> exposure = 1.5
> hue=.1
> 
> learning_rate=0.001
> burn_in=1000
> max_batches=84000
> policy=steps
> steps=75600
> scales=.1
> 
> [convolutional]
> batch_normalize=1
> filters=32
> size=3
> stride=1
> pad=1
> activation=leaky
> 
> # Downsample
> 
> [convolutional]
> batch_normalize=1
> filters=64
> size=3
> stride=2
> pad=1
> activation=leaky
> 
> [convolutional]
> batch_normalize=1
> filters=32
> size=1
> stride=1
> pad=1
> activation=leaky
> 
> [convolutional]
> batch_normalize=1
> filters=64
> size=3
> stride=1
> pad=1
> activation=leaky
> 
[... 703 lines cut ...]
> stride=1
> pad=1
> filters=256
> activation=leaky
> 
> [convolutional]
> size=1
> stride=1
> pad=1
> filters=141
> activation=linear
> 
> [yolo]
> mask = 0,1,2
> anchors = 10,13,  16,30,  33,23,  30,61,  62,45,  59,119,  116,90,  156,198,  373,326
> classes=42
> num=9
> jitter=.3
> ignore_thresh = .5
> truth_thresh = 1
> random=1

Can you see some other problem ? 
Do you need some more details ? 
Thanks for your help!

jkjung-avt, 2020-04-02:
Comparing against the original [yolov3.cfg](https://github.com/pjreddie/darknet/blob/master/cfg/yolov3.cfg), you have made changes to "strides" in the network.  That's why you'd get a different size of feature map as output.

For example, [line 716~717](https://github.com/pjreddie/darknet/blob/master/cfg/yolov3.cfg#L717) of the original yolov3.cfg says:

```
[upsample]
stride=2
```

And you have modified it to:

```
[upsample]
stride=4
```

GuyChriqui, 2020-04-02:
Thanks for your help!
I retrained the model with the original tiny-yolo-416, and its worked !
Thanks!

closed 2020-04-02 by GuyChriqui

jkjung-avt, 2020-06-13:
I've added a "--category_num" command-line option to make it easier to adapt my TensorRT YOLOv3 code to custom trained models.  Please check out my blog post [TensorRT YOLOv3 For Custom Trained Models](https://jkjung-avt.github.io/trt-yolov3-custom/) for details.