jkjung-avt/tensorrt_demos #76: Custom model using ssd-uff

jenabbhatia, 2020-04-01:
hello, jkjung,
      I am using trt_ssd.py file to run face detection model which is trained using ssd_mobilenet_v2. I have made changes in build_engine.py and build_engines.sh for the face. But when I run trt_ssd.py file, I got error as below:
Traceback (most recent call last):
  File "build_engine.py", line 230, in <module>
    main()
  File "build_engine.py", line 207, in main
    spec)
  File "build_engine.py", line 183, in add_plugin
    graph.find_nodes_by_op("NMS_TRT")[0].input.remove("Input")
  File "/home/sensomatic1/sensoEnvironment/lib/python3.6/site-packages/google/protobuf/internal/containers.py", line 296, in remove
    self._values.remove(elem)
ValueError: list.remove(x): x not in list

I am not getting what this error is about. Please help me. Thanks in advance

jkjung-avt, 2020-04-01:
If you want to use TensorRT SSD inference code in this repository without modification, I'd suggest you use the "6518c1c" version of tensorflow object detection API.  More details could be found in the following posts.

https://github.com/jkjung-avt/tensorrt_demos/issues/54#issuecomment-583707548
https://github.com/jkjung-avt/tensorrt_demos/issues/34#issuecomment-570784902

jenabbhatia, 2020-04-01:
Hello,
  Thanks for the reply. But I am confused regarding two things: 
  1.) first I have to use the Tensorflow object detection API of version 6518c1c. I am confused with this version. I cant find this version API.
   2.) Do I need to train my Face detection model again with this API version

Varat7v2, 2020-04-02:
Hi jkjung,
I installed 6518c1c version of TensorFlow Object Detection API and trained a model. Once I exported the frozen graph model, I used your "build_engine.py" for optimized conversion to trt model. Only .uff and .pbtxt files were generated but .bin file was not created. The following error was thrown.

```
[TensorRT] ERROR: UffParser: Validator error: FeatureExtractor/MobilenetV2/layer_19_2_Conv2d_4_3x3_s2_256/BatchNorm/FusedBatchNormV3: Unsupported operation _FusedBatchNormV3
[TensorRT] ERROR: Network must have at least one output
Traceback (most recent call last):
  File "build_engine.py", line 230, in <module>
    main()
  File "build_engine.py", line 224, in main
    buf = engine.serialize()
AttributeError: 'NoneType' object has no attribute 'serialize'
```
Can you please help me with this problem. I have been struggling for more than a week to convert my custom model to trt model.

jkjung-avt, 2020-04-02:
@Varat7v2  I think "FusedBatchNormv3" was introduced in [tensorflow-1.15.0](https://github.com/tensorflow/tensorflow/releases/tag/v1.15.0) and it is not supported by UFF.  You could avoid this problem by exporting the frozen graph using an older version of tensorflow (1.14.x or earlier).

Alternatively, you could use graphsurgeon to replace "FusedBatchNormv3" nodes with "FusedBatchNorm".  For that, you could reference my code here: https://github.com/jkjung-avt/tensorrt_demos/blob/ssd_mobilenet_v3/ssd/build_engine.py#L286

jenabbhatia, 2020-04-02:
@jkjung-avt Thank you so much. Your code is really helpful. I was able to solve my issues with your suggestions. Thank you once again :)

Varat7v2, 2020-04-02:
Thanks [**jkjung-avt**](https://github.com/jkjung-avt). Downgrading the Tensorflow Object Detection API version to 6518c1c and reinstalling the TensorFlow version 1.14 worked for me with my own custom-trained model. 

I have one last question: If I want to this whole procedure but with ONNX instead of UFF, what should I do? In your repository, ONNX conversion code is only for YOLO. I want to do the SSD model for Jetson TX2 with ONNX.

Thank you once more.

jkjung-avt, 2020-04-03:
I haven't tried converting tensorflow models to ONNX and TensorRT engines.  (In fact, I haven't seen NVIDIA providing a good example about how to add TensorRT plugins with ONNX.)

I will try to share some ONNX/TensorRT examples when I get more familiarized with it.

closed 2020-04-03 by jkjung-avt