jkjung-avt/tensorrt_demos #77: Why meet issues after remove "key = cv2.waitKey(1)"

sudo-wang, 2020-04-01:
Hi, @jkjung-avt 
Thanks for your work! It helps me a lot. You are very kind! 
I want to achieve that when the video detection is finished, stop the loop and exit directly. Instead of going through the loop until the 'ESC' key is pressed.
So I calculated the total number of frames in the video file, and remove "key = cv2.waitKey(1)". Just like this:
('total' is the total number of frames in the video file; 
'frameIndex' is a variable of +1 after each loop detection )
[image]
In the end, I run python trt_ssd_async.py --model ssd_mobilenet_v1_coco --file --filename ./examples/test_video/IMG_8706.MP4
But **the log is right, the detected video was not show.**
[image]
When I added "key = cv2.waitKey(1)" again, everything was fine, 
But after the video is finished, it loops again for a while (less than the second time) before it ends. May be the delay caused by "key = cv2.waitkey (1)"?

**I do not know why I remove "key = cv2.waitKey(1)", and the detection video was not show.**
Please give my some hints for modification, Sincere thanks for you!

jkjung-avt, 2020-04-01:
Your question is just basic programming of opencv API.  It should be easy for your to find reference online and do the studying by yourself.  For example, you could reference [this StackOverflow post](https://stackoverflow.com/questions/44854699/why-does-the-cv2-imshow-does-not-render-without-cv2-waitkey/44854945).

sudo-wang, 2020-04-01:
Ok, Thank you all the same

closed 2020-04-01 by sudo-wang