jkjung-avt/tensorrt_demos #78: convert dimension for original video

jeansely, 2020-04-01:
Hi,
Thank you  this demo,

Our video resolution has a high resolution(2048*1536) , which decreases after entering the process of finding objects.
After finding the location of the objects, we will convert the coordinates of the objects to find more details in original image
I was using the following code which was extremely slow to process. Please guide. Thank you

`def loop_and_detect(cam, trt_yolov3, conf_th, vis):
    """Continuously capture images from camera and do object detection.
    # Arguments
      cam: the camera instance (video source).
      trt_yolov3: the TRT YOLOv3 object detector instance.
      conf_th: confidence/score threshold for object detection.
      vis: for visualization.
    """
    full_scrn = False
    fps = 0.0
    tic = time.time()
    while True:
        if cv2.getWindowProperty(WINDOW_NAME, 0) < 0:
            break
        img = cam.read()
        if img is not None:
            boxes, confs, clss = trt_yolov3.detect(img, conf_th)
            img = vis.draw_bboxes(img, boxes, confs, clss)
            img = show_fps(img, fps)
            cv2.imshow(WINDOW_NAME, img)
___________________________________________________________________________    
           
          *original_image = cv2.VideoCapture("rstp:....")
           *  rect_org, fram_org = original_image
            * new_boxes = boxes*3.2 # convert for original image
            * cut bounding box from orginal image

_________________________________________________________________________________

           toc = time.time()
            curr_fps = 1.0 / (toc - tic)
            # calculate an exponentially decaying average of fps number
            fps = curr_fps if fps == 0.0 else (fps*0.95 + curr_fps*0.05)
            tic = toc
        key = cv2.waitKey(1)
        if key == 27:  # ESC key: quit program
            break
        elif key == ord('F') or key == ord('f'):  # Toggle fullscreen
            full_scrn = not full_scrn
            set_display(WINDOW_NAME, full_scrn)`

jkjung-avt, 2020-04-02:
How slow is "extremely slow"?  The TensorRT "yolov3-608" runs at only 1.53 FPS on Jetson Nano.  Maybe you should consider using another object detector than YOLOv3.

jeansely, 2020-04-02:
Hi, I use xavier model with tiny version. And the camera has 2048*1536 pixel. After object detection I have to classify the object and need more pixel of the object.

jkjung-avt, 2020-04-02:
I think your question is not well specified.  As such, I don't know how I could offer some help.

For example,

* What is your targeted frame rate (or inference speed)?  And what frame rate are you getting right now?
* Have you done some profiling of your whole image processing pipeline?  Which stage of the processing would need to be optimized first?
* In your 1st post, you said your modified loop_and_detect loop runs too slowly.  How does it compare to the original yolov3-608 model?

jeansely, 2020-04-02:
I take 38 fps on xavier with tiny version. I think، The best place to add code is loop_and_detect. Because the position of the objects has been achieved and I have to convert these coordinates ( bounding box) to get more pixels than the objects in the input image that have a different resolution.

closed 2020-05-01 by jkjung-avt