jkjung-avt/tensorrt_demos #79: Can I use tensorrt to model other yolo architecture ?

brian208579, 2020-04-01:
Hellow @jkjung-avt 
I using https://github.com/AlexeyAB/darknet training Other architectures of yolo
For example ( figure ) :
[image]
Now, I want to speed up these models on tensorrt and using jetson nano inference
Can I use tensorrt to model other yolo architecture ?
Appreciate

jkjung-avt, 2020-04-02:
I think it's doable.  But you'll likely have to make significant changes to [yolov3_to_onnx.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolov3_onnx/yolov3_to_onnx.py), [onnx_to_tensorrt](https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolov3_onnx/onnx_to_tensorrt.py) and [yolov3.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/yolov3.py).

brian208579, 2020-04-06:
@jkjung-avt 
ok, I will try it, thank you very much ~

closed 2020-04-06 by brian208579