jkjung-avt/tensorrt_demos #80: AttributeError: 'Nonetype' object has no attribute 'serialize'

gpqls, 2020-04-06:
This error occurred while I was performing `build_engines.py ${model}`

gpqls, 2020-04-06:
python3 build_engine.py ssd_mobilenet_v2_face
2020-04-06 16:44:47.936106: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcudart.so.10.0
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/uff/converters/tensorflow/conversion_helpers.py:18: The name tf.GraphDef is deprecated. Please use tf.compat.v1.GraphDef instead.

WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/graphsurgeon/_utils.py:2: The name tf.NodeDef is deprecated. Please use tf.compat.v1.NodeDef instead.

[TensorRT] INFO: Plugin Creator registration succeeded - GridAnchor_TRT
[TensorRT] INFO: Plugin Creator registration succeeded - NMS_TRT
[TensorRT] INFO: Plugin Creator registration succeeded - Reorg_TRT
[TensorRT] INFO: Plugin Creator registration succeeded - Region_TRT
[TensorRT] INFO: Plugin Creator registration succeeded - Clip_TRT
[TensorRT] INFO: Plugin Creator registration succeeded - LReLU_TRT
[TensorRT] INFO: Plugin Creator registration succeeded - PriorBox_TRT
[TensorRT] INFO: Plugin Creator registration succeeded - Normalize_TRT
[TensorRT] INFO: Plugin Creator registration succeeded - RPROI_TRT
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/graphsurgeon/StaticGraph.py:123: FastGFile.__init__ (from tensorflow.python.platform.gfile) is deprecated and will be removed in a future version.
Instructions for updating:
Use tf.gfile.GFile.
WARNING: To create TensorRT plugin nodes, please use the `create_plugin_node` function instead.
UFF Version 0.5.5
=== Automatically deduced input nodes ===
[name: "Input"
op: "Placeholder"
input: "Cast"
attr {
  key: "dtype"
  value {
    type: DT_FLOAT
  }
}
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: 1
      }
      dim {
        size: 3
      }
      dim {
        size: 300
      }
      dim {
        size: 300
      }
    }
  }
}
]
=========================================

Using output node NMS
Converting to UFF graph
Warning: No conversion function registered for layer: NMS_TRT yet.
Converting NMS as custom op: NMS_TRT
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:189: The name tf.AttrValue is deprecated. Please use tf.compat.v1.AttrValue instead.

Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_conf as custom op: FlattenConcat_TRT
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/layer_19_2_Conv2d_5_3x3_s2_128/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/layer_19_2_Conv2d_5_3x3_s2_128_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/layer_19_1_Conv2d_5_1x1_64/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/layer_19_2_Conv2d_4_3x3_s2_256/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/layer_19_2_Conv2d_4_3x3_s2_256_depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
[... 110 lines cut ...]
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/expanded_conv_14/depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/expanded_conv_14/expand/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/expanded_conv_15/project/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/expanded_conv_15/depthwise/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: FusedBatchNormV3 yet.
Converting FeatureExtractor/MobilenetV2/expanded_conv_15/expand/BatchNorm/FusedBatchNormV3 as custom op: FusedBatchNormV3
Warning: No conversion function registered for layer: GridAnchor_TRT yet.
Converting MultipleGridAnchorGenerator as custom op: GridAnchor_TRT
Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_loc as custom op: FlattenConcat_TRT
No. nodes: 676
UFF Output written to /home/hyebin/tensorrt_demos/ssd/tmp_v2_face.uff
UFF Text Output written to /home/hyebin/tensorrt_demos/ssd/tmp_v2_face.pbtxt
[TensorRT] ERROR: UFFParser: Validator error: FeatureExtractor/MobilenetV2/layer_19_2_Conv2d_5_3x3_s2_128_depthwise/BatchNorm/FusedBatchNormV3: Unsupported operation _FusedBatchNormV3
[TensorRT] ERROR: Network must have at least one output
Traceback (most recent call last):
  File "build_engine.py", line 230, in <module>
    main()
  File "build_engine.py", line 224, in main
    buf = engine.serialize()
AttributeError: 'NoneType' object has no attribute 'serialize'

gpqls, 2020-04-06:
There was an error, but the tmp_v2_face.pbtxt file, uff file, and bin file were created.
The files are probably not normal because an error has occurred.
I want to know the cause of the error.

jkjung-avt, 2020-04-06:
https://github.com/jkjung-avt/tensorrt_demos/issues/76#issuecomment-607879831

gpqls, 2020-04-06:
I use tensorflow 1.14.0 and add 
`def replace_fusedbnv3(graph):
    """Replace all 'FusedBatchNormV3' in the graph with 'FusedBatchNorm'.
    NOTE: 'FusedBatchNormV3' is not supported by UFF parser.
    https://devtalk.nvidia.com/default/topic/1066445/tensorrt/tensorrt-6-0-1-tensorflow-1-14-no-conversion-function-registered-for-layer-fusedbatchnormv3-yet/post/5403567/#5403567
    """
    for node in graph.find_nodes_by_op('FusedBatchNormV3'):
        gs.update_node(node, op='FusedBatchNorm')
    return graph`

but the same error keeps coming out.

gpqls, 2020-04-06:
That error was corrected by adding a function to te main part, but another error occurred.
`AttributeError: module 'graphsurgeon' has no attribute 'update_node'
`

gpqls, 2020-04-06:
`Traceback (most recent call last):
  File "build_engine.py", line 240, in <module>
    main()
  File "build_engine.py", line 217, in main
    dynamic_graph = replace_fusedbnv3(dynamic_graph)
  File "build_engine.py", line 195, in replace_fusedbnv3
    gs.update_node(node, op='FusedBatchNorm')
AttributeError: module 'graphsurgeon' has no attribute 'update_node'
`

jkjung-avt, 2020-04-06:
Ah, I think I tested "replace_fusedbnv3()" with TensorRT 7.  If you are using a different version of TensorRT, you need to figure out by yourself which graphsurgeon API you could use.

As I stated in https://github.com/jkjung-avt/tensorrt_demos/issues/76#issuecomment-607879831, alternatively you could avoid this problem by re-exporting the frozen inference graph with tensorflow-1.14.x or lower.

gpqls, 2020-04-06:
I am using tensorflow 1.14 and using this version is the result of exporting a fixed inference graph.

I don't know exactly what it means to figure out graphsurgeon api.

jkjung-avt, 2020-04-06:
It's strange.  If you refer to https://github.com/jkjung-avt/tensorrt_demos/issues/76#issuecomment-608041223, you could see other people (previously using tensorflow-1.15.0) confirmed that using tensorflow-1.14.0 solved the exact problem.

I myself have mostly used tensorflow-1.12.2.

The graphsurgeon approach would require you to read graphsurgeon source code and come up the solution by yourself.  The update_node() API is not formally documented ([link](https://docs.nvidia.com/deeplearning/sdk/tensorrt-api/python_api/graphsurgeon/graphsurgeon.html)).  I found it by reading the source code...

closed 2020-05-04 by gpqls