jkjung-avt/tensorrt_demos #83: Errors in deserializing plugin FlattenConcat_TRT

sanjain2004, 2020-04-08:
We have a model based on SSD and your code to serialize it using python. 

In my C++ code I am deserializing the model.

I get these errors:
WARNING: Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors. 
ERROR: getPluginCreator could not find plugin FlattenConcat_TRT version 1 namespace  
ERROR: Cannot deserialize plugin FlattenConcat_TRT 
ERROR: getPluginCreator could not find plugin FlattenConcat_TRT version 1 namespace  
ERROR: Cannot deserialize plugin FlattenConcat_TRT

I took your libflattenconcat.so.5 and libflattenconcat.so.6 and tried with both. I still get the same error.

Thanks

jkjung-avt, 2020-04-08:
I'm not sure whether you are using TensorRT 5 or TensorRT 6.  I use TensorRT 5 as an example.

Please download TensorRT-5.1.5.0 release from NVIDIA web site.  For example, download this file (login required): https://developer.nvidia.com/compute/machine-learning/tensorrt/5.1/ga/tars/TensorRT-5.1.5.0.Ubuntu-18.04.2.x86_64-gnu.cuda-10.0.cudnn7.5.tar.gz

After extracting the TensorRT package, you could refer to these 2 samples:

* samples/python/uff_ssd/ -> you could find source code of "libflattenconcat.so" in this sample under the "plugin" directory.
* samples/sampleUffSSD/

I guess the easiest solution for you is to refer to samples/sampleUffSSD/sampleUffSSD.cpp and copy over all code related to "class FlattenConcat" and "class FlattenConcatPluginCreator".

Otherwise, you could try to link the correct version of "libflattenconcat.so" with your code (and may also add some code to register "FlattenConcatPluginCreator").

sanjain2004, 2020-04-08:
I have 5.0.6. The source for flattenConcat uses IPluginV2Ext, but version 5.0.6 does not have it. Any suggestions on where to get the Plugin source for 5.0.6?

jkjung-avt, 2020-04-09:
You could download TensorRT-5.0.2.6 GA release from this page (login required): https://developer.nvidia.com/nvidia-tensorrt-5x-download

closed 2020-04-15 by jkjung-avt