jkjung-avt/tensorrt_demos #85: ValueError: cannot reshape array of size 4 into shape (10,2,1,1)

jeansely, 2020-04-11:
Hi, 
Thank you for this code:
my input images have resolution 320*64 (car plate) and I  change dimension but in yolov3.py take below error:
(i have one class and change 255 to 18 and category_num=1)

  File "/home/cv/Downloads/tensorrt_demos-master/tensorrt_demos-master/utils/yolov3.py", line 486, in detect
    trt_outputs, shape_orig_WH, conf_th)
  File "/home/cv/Downloads/tensorrt_demos-master/tensorrt_demos-master/utils/yolov3.py", line 112, in process
    outputs_reshaped, resolution_raw, conf_th)
  File "/home/cv/Downloads/tensorrt_demos-master/tensorrt_demos-master/utils/yolov3.py", line 162, in _process_yolo_output
    box, category, confidence = self._process_feats(output, mask)
  File "/home/cv/Downloads/tensorrt_demos-master/tensorrt_demos-master/utils/yolov3.py", line 234,  in _process_feats
    col = col.reshape(grid_h, grid_w, 1, 1).repeat(3, axis=-2)
ValueError: cannot reshape array of size 4 into shape (10,2,1,1)

Thank you,

jkjung-avt, 2020-04-12:
Please provide your yolov3.cfg so I can take a look.

jkjung-avt, 2020-04-23:
https://github.com/jkjung-avt/tensorrt_demos/issues/95#issuecomment-616424642

closed 2020-04-23 by jkjung-avt