jkjung-avt/tensorrt_demos #87: libflattenconcat.so file

SahilChachere, 2020-04-13:
what is use of libflattenconcat.so, 5, 6 files in jetson nano project, and we use only one file amongst 3.

jkjung-avt, 2020-04-13:
That is the pre-compiled library for the "FlattenConcat_TRT" plugin.  The original NVIDIA repository also has such a file: https://github.com/AastaNV/TRT_object_detection/blob/master/lib/libflattenconcat.so

The .5 and .6 files are for TensorRT 5 and TensorRT 6 respectively.  In [ssd/install.sh](https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/install.sh#L28) I make s symbolic link depending on which version of TensorRT is used in the system.

closed 2020-04-15 by jkjung-avt

nhphuong91, 2020-11-26:
@jkjung-avt How can I build the pre-compiled library for the "FlattenConcat_TRT" plugin? I want to make the TensorRT 7 version of it. Thanks in advanced!

jkjung-avt, 2020-11-26:
@nhphuong91  The "FlattenConcat_TRT" has been included in the TensorRT 7 libraries already.  You don't need to re-build it for TensorRT 7.