jkjung-avt/tensorrt_demos #89: run two or more engine on a tensorrt

jeansely, 2020-04-14:
Hi,
how can i  run two engine (two mode of yolov3) into a jetson(concurrently) , models are different.
i used a form that had this error on add memory:

boxes_2, confs_2, clss_2 = trt_yolov3.detect(img, conf_th)
  File "/home/cv/Downloads/tensorrt_demos-master/tensorrt_demos-master/utils2/yolov3.py", line 479, in detect
    stream=self.stream)
  File "/home/cv/Downloads/tensorrt_demos-master/tensorrt_demos-master/utils2/yolov3.py", line 367, in do_inference
    [cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
  File "/home/cv/Downloads/tensorrt_demos-master/tensorrt_demos-master/utils2/yolov3.py", line 367, in <listcomp>
    [cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument

Thank you.

jkjung-avt, 2020-04-14:
I haven't tried that before.  Maybe you could try to create 2 cuda contexts for the 2 engines, and manually switch cuda context before calling do_inference() of the corresponding engine.  Reference: https://github.com/jkjung-avt/tensorrt_demos/issues/67#issuecomment-604887874

jeansely, 2020-04-14:
1. In your code , how should I distinguish  between two engines  or contexts? for example:
 syntax: return self.engine.create_execution_context(?????) or  for input and output & engines.

2. if i define two classes, class TrtYOLOv3_1 and class TrtYOLOv3_2  when determinate(initialize) scale of memory device_mem = cuda.mem_alloc(host_mem.nbytes) , allocate_buffers(self.engine), and    [cuda.memcpy_dtoh_async(out.host, out.device, stream) for out in outputs].
Does not create any conflicts on memory

Many Thanks.

jkjung-avt, 2020-04-15:
Sorry, I don't have a good answer now.  I myself am still learning these stuffs.  I'll get back to you if I find answers later on.

jeansely, 2020-04-28:
Hi,
I tested it and it is possible.
the error:
"pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument"
It is related to the non-uniform dimensions of the given training model and the operational model.

jkjung-avt, 2020-04-29:
I'm glad to head this update.

closed 2020-05-01 by jkjung-avt

jeansely, 2020-05-03:
Hi,
In order to run multiple model with TensorRT, NVIDIA  recommend to either use NVIDIA "deepstream" or NVIDIA Triton Inference Server.
but i load(call) two engine(.trt) into  your  code  . it work very good

jkjung-avt, 2020-08-16:
https://forums.developer.nvidia.com/t/unable-to-run-two-tensorrt-models-in-a-cascade-manner/145274/5?u=jkjung13