jkjung-avt/tensorrt_demos #90: ERROR: Could not register plugin creator  GridAnchor_TRT Namespace

sanjain2004, 2020-04-15:
I am running TensorRT with C++ libraries. I have linked nvinfer and nvinfer_plugin libraries. 
I have tried it with static and dynamic libraries.

When I do an objdump of the shared objects under /usr/lib/aarch64-linux-gnu, I can see the plugins there. 

My code starts with a call to initLibNvInferPlugins(&gLogger1, "");

Immediately after the call I read the serialized model and call deserialize on it. After that I see the following errors:
ERROR: Could not register plugin creator  GridAnchor_TRT Namespace 
ERROR: Could not register plugin creator  NMS_TRT Namespace 
ERROR: Could not register plugin creator  Reorg_TRT Namespace 
ERROR: Could not register plugin creator  Region_TRT Namespace 
ERROR: Could not register plugin creator  Clip_TRT Namespace 
ERROR: Could not register plugin creator  LReLU_TRT Namespace 
ERROR: Could not register plugin creator  PriorBox_TRT Namespace 
ERROR: Could not register plugin creator  Normalize_TRT Namespace 
ERROR: Could not register plugin creator  RPROI_TRT Namespace 
ERROR: Could not register plugin creator  GridAnchor_TRT Namespace 
ERROR: Could not register plugin creator  NMS_TRT Namespace 
ERROR: Could not register plugin creator  Reorg_TRT Namespace 
ERROR: Could not register plugin creator  Region_TRT Namespace 
ERROR: Could not register plugin creator  Clip_TRT Namespace 
ERROR: Could not register plugin creator  LReLU_TRT Namespace 
ERROR: Could not register plugin creator  PriorBox_TRT Namespace 
ERROR: Could not register plugin creator  Normalize_TRT Namespace 
ERROR: Could not register plugin creator  RPROI_TRT Namespace 

On the same jetson nano, the same serialized model works in Python. 

Any help is appreciated.

Thanks

jkjung-avt, 2020-04-16:
From the description, you seem doing it correctly.  I'm not able to guess what the problem is.  I think I need take a look at your source code to see if I could spot something.

jkjung-avt, 2020-05-01:
If you could post your source code, I will take a look.  Otherwise, I will close this issue for now.

closed 2020-05-01 by sanjain2004