jkjung-avt/tensorrt_demos #91: Gstream

PythonImageDeveloper, 2020-04-15:
Hi,
When I install opencv with enabled GStreamer on jetson nano, It's a automatically use H264 Hardware accelarator decoder? Is FFMPEG also use H264 HW decoder?

jkjung-avt, 2020-04-16:
You have to explicitly tell OpenCV to use the hardware H.264 decoder by putting the "omxh264dec" element in the gstreamer pipeline, as exampled in my [tegra-cam.py](https://gist.github.com/jkjung-avt/86b60a7723b97da19f7bfa3cb7d2690e#file-tegra-cam-py-L55) script.

According to [NVIDIA's documentation](https://elinux.org/Jetson/H264_Codec), using the gstreamer element is the only way to access the hardware H.264 codec on the Jetson platforms.  I'm not sure whether FFMPEG could use the gstreamer pipeline as the underlying video decoding mechanism (most likely not).  So I would suggest you to use gstreamer.

PythonImageDeveloper, 2020-04-16:
When I install opencv from source with cuda and gstream enabled, opencv compile gstream during instalization, I want to know Is it use last version of gstream or is used related version of gstream for opencv? 
Assme the last version of gstream is 1.31, when install opencv 3.x and opencv 4.x, are both opencv version use the last version of gstream?

jkjung-avt, 2020-04-16:
I just used the version which gets installed with "sudo apt install".  Please refer to my blog post: https://jkjung-avt.github.io/opencv-on-nano/

jkjung-avt, 2020-04-23:
Closing this issue since it is not related to code in this repository.

closed 2020-04-23 by jkjung-avt